XCSoar/XCSoar #64: `android/ioio` git submodule pointing to broken `git.xcsoar.org` server

Turbo87, 2018-02-14:
It seems that git.xcsoar.org is currently not or no longer available, but the `android/ioio` submodule is still pointing to it on both branches, which currently makes this project (and others depending on it) impossible to build.

/cc @MaxKellermann

Turbo87, 2018-02-20:
I've forked https://github.com/ytai/ioio to https://github.com/XCSoar/ioio, but it seems that the referenced commits from the submodule are not part of the official repository.

@felixhaedicke @MaxKellermann @lordfolken @TobiasLohner does one of you still have the `android/ioio` repository locally and can update https://github.com/XCSoar/ioio to include the missing commits?

kedder, 2018-02-20:
Not me, unfortunately. I recently forked XCSoar, but didn't pull git modules.

Turbo87, 2018-02-23:
I have received a recent enough copy from @M-Scholli that includes the missing commits and pushed it to https://github.com/XCSoar/ioio now. I've opened #65 and #66 to fix the broken submodule URLs.

PR #65 (2018-03-02): Adjust URL for `android/ioio` submodule
Resolves #64
closed 2018-03-02 by Turbo87