XCSoar/XCSoar #69: Keyhole for AAT task

Exadios, 2018-03-07:
This issue replicates [Keyhole for AAT task](https://bugs.xcsoar.org/ticket/1687) here on github.

kerel-fs, 2019-09-18:
#148 addresses this issue.

Exadios, 2019-09-19:
As I messaged you previously I have your branch on this subject. I've not yet had a chance to have a close look, test, etc. But it is on my list of things to do.

Since it's mostly addressed, I'm going to close this issue.

Exadios, 2019-09-19:
To be addressed by the pull request that will, soon, replace [url=https://github.com/XCSoar/XCSoar/pull/148]#148[/url].

kerel-fs, 2019-09-19:
Great! Note that you don't have to create a new Pull Request but can force-push the updated commits to the original `Exadios:ticket_1687_on_master` branch and re-open the old Pull Request. This way we don't "loose" the initial discussion which should simplify review.

Exadios, 2019-09-20:
OK, Thanks for the tip.

Exadios, 2019-10-03:
> Great! Note that you don't have to create a new Pull Request but can force-push the updated commits to the original `Exadios:ticket_1687_on_master` branch and re-open the old Pull Request. This way we don't "loose" the initial discussion which should simplify review.

So, how do I re-open the Pull Request? I have pushed the branch but can see no way to re-open. I suspect I do not have sufficient privilege to do this.

kerel-fs, 2019-10-03:
Indeed, seems that you need special privileges to re-open the Pull Request (on the other hand, you can close it yourself, so I don't fully understand the underlying logic yet).

So, I'll re-open it.

kerel-fs, 2019-10-03:
Oops, didn't re-open the PR but this issue instead. I've as well no permission to re-open #148. Probably you should comment there and ask cirrus to re-open the PR.

Exadios, 2019-10-07:
OK. Thanks.

lordfolken, 2020-05-04:
Are you guys going to submit a new PR?

lordfolken, 2022-08-23:
There is a new ticket #938 does this cover this here?

lordfolken, 2022-09-05:
#938 implements this.

closed 2022-09-05 by lordfolken