XCSoar/XCSoar #70: Simplify AAT turnpoint implementation.

Exadios, 2018-03-07:
This issue replicates [Ticket 3844](https://bugs.xcsoar.org/ticket/3844) on github.

kerel-fs, 2019-09-18:
@Exadios Can you define the issue more precisely?

Unfortunately the bug tracker is gone (there are backups, but it's unclear if/when it will be back online).

Exadios, 2019-09-19:
So, originally, the idea was to get rid of the specialized AAT turnpoint types and re implement them as specific parameterizations of what I have called 'VariableKeyholeZone' (which would have been renamed to somthing more appropriate). So, the simplification would be to the code - getting rid of the numerous 'case' structures, reworking the factories etc.

Later, after writing this issue, and after talking to some club members who are into the competition scene, it occurred to me that it might be useful to add a new task type - the SeeYou task type. The idea here would be to simplify the data entry from the SeeYou task sheet into XCSoar.

The thing is that I'm busy doing other things and will probably never implement this. And nobody else has shown interest so I'm going to close this issue.

closed 2019-09-19 by Exadios