XCSoar/XCSoar #76: assert in OrderedTask

lordfolken, 2018-05-07:
xcsoar: ./src/Engine/Task/Ordered/OrderedTask.hpp:361: const TaskProjection& OrderedTask::GetTaskProjection() const: Assertion `!IsEmpty()' failed.
in @e0418b0e40d8b98329c290dbeb0a8eafac8c696a

lordfolken, 2018-05-07:
This happens also with a blank .xcsoar directory.

Running in sim mode, about 4 seconds after starting the plane in one direction.

lordfolken, 2018-05-07:
build command is make -j5 TARGET=UNIX

lordfolken, 2018-05-07:
strace output: 
clock_gettime(CLOCK_MONOTONIC, {tv_sec=11904701, tv_nsec=948718708}) = 0
clock_gettime(CLOCK_MONOTONIC, {tv_sec=11904701, tv_nsec=948735880}) = 0
poll([{fd=14, events=POLLIN|POLLPRI}], 1, 1000) = 1 ([{fd=14, revents=POLLIN|POLLPRI}])
ioctl(14, _IOC(_IOC_READ|_IOC_WRITE, 0x46, 0x52, 0x10), 0x7ffd5fca1d20) = 0
clock_gettime(CLOCK_MONOTONIC, {tv_sec=11904701, tv_nsec=948800300}) = 0
clock_gettime(CLOCK_MONOTONIC, {tv_sec=11904701, tv_nsec=948816790}) = 0
clock_gettime(CLOCK_MONOTONIC, {tv_sec=11904701, tv_nsec=948833151}) = 0
clock_gettime(CLOCK_MONOTONIC, {tv_sec=11904701, tv_nsec=948849604}) = 0
clock_gettime(CLOCK_MONOTONIC, {tv_sec=11904701, tv_nsec=948865860}) = 0
clock_gettime(CLOCK_MONOTONIC, {tv_sec=11904701, tv_nsec=948882327}) = 0
clock_gettime(CLOCK_MONOTONIC, {tv_sec=11904701, tv_nsec=948901347}) = 0
sched_yield()                           = 0
sched_yield()                           = 0
clock_gettime(CLOCK_MONOTONIC, {tv_sec=11904701, tv_nsec=948955950}) = 0
sched_yield()                           = 0
clock_gettime(CLOCK_MONOTONIC, {tv_sec=11904701, tv_nsec=948985745}) = 0
clock_gettime(CLOCK_MONOTONIC, {tv_sec=11904701, tv_nsec=949000492}) = 0
clock_gettime(CLOCK_MONOTONIC, {tv_sec=11904701, tv_nsec=949015019}) = 0
clock_gettime(CLOCK_MONOTONIC, {tv_sec=11904701, tv_nsec=949030122}) = 0
clock_gettime(CLOCK_MONOTONIC, {tv_sec=11904701, tv_nsec=949044929}) = 0
sched_yield()                           = 0
epoll_wait(4, [{EPOLLIN, {u32=2438849588, u64=94010683024436}}], 128, -1) = 1
timerfd_settime(5, 0, {it_interval={tv_sec=0, tv_nsec=0}, it_value={tv_sec=0, tv_nsec=414595000}}, {it_interval={tv_sec=0, tv_nsec=0}, it_value={tv_sec=0, tv_nsec=0}}) = 0
epoll_wait(4, [{EPOLLIN, {u32=2438849588, u64=94010683024436}}], 128, -1) = 1
timerfd_settime(5, 0, {it_interval={tv_sec=0, tv_nsec=0}, it_value={tv_sec=300, tv_nsec=0}}, {it_interval={tv_sec=0, tv_nsec=0}, it_value={tv_sec=0, tv_nsec=0}}) = 0
timerfd_settime(5, 0, {it_interval={tv_sec=0, tv_nsec=0}, it_value={tv_sec=0, tv_nsec=499992000}}, {it_interval={tv_sec=0, tv_nsec=0}, it_value={tv_sec=299, tv_nsec=999781856}}) = 0
epoll_wait(4, [{EPOLLIN, {u32=2438849588, u64=94010683024436}}], 128, -1) = 1
timerfd_settime(5, 0, {it_interval={tv_sec=0, tv_nsec=0}, it_value={tv_sec=300, tv_nsec=0}}, {it_interval={tv_sec=0, tv_nsec=0}, it_value={tv_sec=0, tv_nsec=0}}) = 0
futex(0x558091baabec, FUTEX_WAKE_OP_PRIVATE, 1, 1, 0x558091baabe8, FUTEX_OP_SET<<28|0<<12|FUTEX_OP_CMP_GT<<24|0x1) = 1
futex(0x558091baab20, FUTEX_WAKE_PRIVATE, 1) = 1
timerfd_settime(5, 0, {it_interval={tv_sec=0, tv_nsec=0}, it_value={tv_sec=0, tv_nsec=499997000}}, {it_interval={tv_sec=0, tv_nsec=0}, it_value={tv_sec=299, tv_nsec=999791974}}) = 0
epoll_wait(4, [{EPOLLIN, {u32=2438914672, u64=94010683089520}}], 128, -1) = 1
epoll_wait(4, [{EPOLLIN, {u32=2438914672, u64=94010683089520}}], 128, -1) = 1
timerfd_settime(5, 0, {it_interval={tv_sec=0, tv_nsec=0}, it_value={tv_sec=0, tv_nsec=24997000}}, {it_interval={tv_sec=0, tv_nsec=0}, it_value={tv_sec=0, tv_nsec=439678541}}) = 0
getpid()                                = 3637
getpid()                                = 3637
clock_gettime(CLOCK_MONOTONIC, {tv_sec=11904702, tv_nsec=949509116}) = 0
clock_gettime(CLOCK_MONOTONIC, {tv_sec=11904702, tv_nsec=949539851}) = 0
poll([{fd=14, events=POLLIN|POLLPRI}], 1, 1000) = 1 ([{fd=14, revents=POLLIN|POLLPRI}])
ioctl(14, _IOC(_IOC_READ|_IOC_WRITE, 0x46, 0x52, 0x10), 0x7ffd5fca1d20) = 0
clock_gettime(CLOCK_MONOTONIC, {tv_sec=11904702, tv_nsec=949628409}) = 0
clock_gettime(CLOCK_MONOTONIC, {tv_sec=11904702, tv_nsec=949651970}) = 0
clock_gettime(CLOCK_MONOTONIC, {tv_sec=11904702, tv_nsec=949675406}) = 0
clock_gettime(CLOCK_MONOTONIC, {tv_sec=11904702, tv_nsec=949698773}) = 0
clock_gettime(CLOCK_MONOTONIC, {tv_sec=11904702, tv_nsec=949721670}) = 0
clock_gettime(CLOCK_MONOTONIC, {tv_sec=11904702, tv_nsec=949744563}) = 0
clock_gettime(CLOCK_MONOTONIC, {tv_sec=11904702, tv_nsec=949772250}) = 0
sched_yield()                           = 0
sched_yield()                           = 0
clock_gettime(CLOCK_MONOTONIC, {tv_sec=11904702, tv_nsec=949852167}) = 0
sched_yield()                           = 0
clock_gettime(CLOCK_MONOTONIC, {tv_sec=11904702, tv_nsec=949898945}) = 0
clock_gettime(CLOCK_MONOTONIC, {tv_sec=11904702, tv_nsec=949921717}) = 0
clock_gettime(CLOCK_MONOTONIC, {tv_sec=11904702, tv_nsec=949944691}) = 0
clock_gettime(CLOCK_MONOTONIC, {tv_sec=11904702, tv_nsec=949968164}) = 0
clock_gettime(CLOCK_MONOTONIC, {tv_sec=11904702, tv_nsec=949991021}) = 0
sched_yield()                           = 0
epoll_wait(4, [{EPOLLIN, {u32=2438849588, u64=94010683024436}}], 128, -1) = 1
timerfd_settime(5, 0, {it_interval={tv_sec=0, tv_nsec=0}, it_value={tv_sec=0, tv_nsec=414649000}}, {it_interval={tv_sec=0, tv_nsec=0}, it_value={tv_sec=0, tv_nsec=0}}) = 0
epoll_wait(4, [{EPOLLIN, {u32=2438849588, u64=94010683024436}}], 128, -1) = 1
timerfd_settime(5, 0, {it_interval={tv_sec=0, tv_nsec=0}, it_value={tv_sec=300, tv_nsec=0}}, {it_interval={tv_sec=0, tv_nsec=0}, it_value={tv_sec=0, tv_nsec=0}}) = 0
timerfd_settime(5, 0, {it_interval={tv_sec=0, tv_nsec=0}, it_value={tv_sec=0, tv_nsec=499993000}}, {it_interval={tv_sec=0, tv_nsec=0}, it_value={tv_sec=299, tv_nsec=999803103}}) = 0
epoll_wait(4, [{EPOLLIN, {u32=2438849588, u64=94010683024436}}], 128, -1) = 1
[... cut ...]
mmap(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7f021ac3c000
rt_sigprocmask(SIG_UNBLOCK, [ABRT], NULL, 8) = 0
rt_sigprocmask(SIG_BLOCK, ~[RTMIN RT_1], [INT TERM], 8) = 0
getpid()                                = 3637
gettid()                                = 3637
tgkill(3637, 3637, SIGABRT)             = 0
rt_sigprocmask(SIG_SETMASK, [INT TERM], NULL, 8) = 0
--- SIGABRT {si_signo=SIGABRT, si_code=SI_TKILL, si_pid=3637, si_uid=1000} ---
+++ killed by SIGABRT +++
[1]    3635 abort      strace ./output/UNIX/bin/xcsoar

kerel-fs, 2018-06-10:
Reproduced this issue with current master, 4c6e754af4.

Steps to reproduce (as already explained by Felix):
Start xcsoar, choose SIM mode, start the plane by dragging it in on direction, wait a few seconds, crash.

kerel-fs, 2018-06-10:
Backtrace using gdb:
```
kerel@fornax:~/Fliegen/xcsoar/xcsoar master(1)+ ± gdb -ex "source tools/gdb.py" output/UNIX/bin/xcsoar
GNU gdb (Debian 7.12-6) 7.12.0.20161007-git
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from output/UNIX/bin/xcsoar...done.
(gdb) r
Starting program: /home/kerel/Fliegen/xcsoar/xcsoar/output/UNIX/bin/xcsoar 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
Free storage 0 KB
Starting XCSoar XCSoar v7.0_preview11-Linux~git#3c472fb1c9
[New Thread 0x7fffede4e700 (LWP 22276)]
Display dpi=95,95
OpenGL: npot=1 vbo=1 fbo=1 stencil=0x84f9
Initialise fonts
Startup dialog
Loading profiles
Failed to load profile: Failed to open /home/kerel//.xcsoar/default.prf: No such file or directory
Loading input events file
OpenTerrain
Loading Topography File...
ReadWaypoints
Failed to read waypoint file: /home/kerel//.xcsoar/user.cup
RASP load
No rasp data file: Failed to open ZIP archive /home/kerel//.xcsoar/xcsoar-rasp.dat
ReadAirspace
Register serial devices
ProgramStarted
[New Thread 0x7fffe9815700 (LWP 22286)]
[New Thread 0x7fffe9014700 (LWP 22287)]
Free storage 14888676 KB
cannot open /home/kerel//.xcsoar/lua/init.lua: No such file or directory
[New Thread 0x7fffe8813700 (LWP 22292)]
xcsoar: ./src/Engine/Task/Ordered/OrderedTask.hpp:361: const TaskProjection& OrderedTask::GetTaskProjection() const: Assertion `!IsEmpty()' failed.

Thread 1 "xcsoar" received signal SIGABRT, Aborted.
__GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
51	../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) bt
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
#1  0x00007ffff522542a in __GI_abort () at abort.c:89
#2  0x00007ffff521ce67 in __assert_fail_base (fmt=<optimized out>, assertion=assertion@entry=0x5555559bbdcc "!IsEmpty()", 
    file=file@entry=0x5555559b4e80 "./src/Engine/Task/Ordered/OrderedTask.hpp", line=line@entry=361, 
    function=function@entry=0x5555559c0400 <OrderedTask::GetTaskProjection() const::__PRETTY_FUNCTION__> "const TaskProjection& OrderedTask::GetTaskProjection() const") at assert.c:92
#3  0x00007ffff521cf12 in __GI___assert_fail (assertion=0x5555559bbdcc "!IsEmpty()", 
    file=0x5555559b4e80 "./src/Engine/Task/Ordered/OrderedTask.hpp", line=361, 
    function=0x5555559c0400 <OrderedTask::GetTaskProjection() const::__PRETTY_FUNCTION__> "const TaskProjection& OrderedTask::GetTaskProjection() const") at assert.c:101
#4  0x0000555555657170 in OrderedTask::GetTaskProjection (this=0x5555564f6dc0) at ./src/Engine/Task/Ordered/OrderedTask.hpp:361
#5  0x00005555556a5bca in MapWindow::DrawTask (this=0x5555564d5fe0, canvas=...) at src/MapWindow/MapWindowTask.cpp:67
#6  0x00005555556a4aef in MapWindow::Render (this=0x5555564d5fe0, canvas=..., rc=...) at src/MapWindow/MapWindowRender.cpp:246
#7  0x00005555556ac27d in GlueMapWindow::Render (this=0x5555564d5fe0, canvas=..., rc=...) at src/MapWindow/GlueMapWindowEvents.cpp:455
#8  0x00005555556a1570 in MapWindow::OnPaintBuffer (this=0x5555564d5fe0, canvas=...) at src/MapWindow/MapWindow.cpp:140
#9  0x00005555556abfd3 in GlueMapWindow::OnPaintBuffer (this=0x5555564d5fe0, canvas=...) at src/MapWindow/GlueMapWindowEvents.cpp:410
#10 0x0000555555873e5f in BufferWindow::OnPaint (this=0x5555564d5fe0, canvas=...) at src/Screen/BufferWindow.cpp:78
#11 0x00005555556a292b in MapWindow::OnPaint (this=0x5555564d5fe0, canvas=...) at src/MapWindow/MapWindowEvents.cpp:94
#12 0x00005555556abf65 in GlueMapWindow::OnPaint (this=0x5555564d5fe0, canvas=...) at src/MapWindow/GlueMapWindowEvents.cpp:392
#13 0x000055555588a4da in WindowList::Paint (this=0x555555f20bc8, canvas=...) at src/Screen/Custom/WList.cpp:234
#14 0x000055555588ba80 in ContainerWindow::OnPaint (this=0x555555f20b90, canvas=...) at src/Screen/Custom/ContainerWindow.cpp:220
#15 0x00005555556f334a in MainWindow::OnPaint (this=0x555555f20b90, canvas=...) at src/MainWindow.cpp:719
#16 0x000055555588c5e3 in TopWindow::Expose (this=0x555555f20b90) at src/Screen/Custom/TopWindow.cpp:105
#17 0x000055555588c64b in TopWindow::Refresh (this=0x555555f20b90) at src/Screen/Custom/TopWindow.cpp:131
#18 0x0000555555899afb in EventLoop::Get (this=0x7fffffffe0c0, event=...) at src/Event/Poll/Loop.cpp:41
#19 0x000055555588faa1 in TopWindow::RunEventLoop (this=0x555555f20b90) at src/Screen/FB/TopWindow.cpp:183
#20 0x00005555557f16b0 in Main () at src/XCSoar.cpp:122
#21 0x00005555557f17e2 in main (argc=1, argv=0x7fffffffe258) at src/XCSoar.cpp:170
```

kerel-fs, 2018-06-10:
Issue also affects 3f2d8dd6a2 / v7.0_preview11  on UNIX, but I can't reproduce on ANDROID / (v7.0-preview11-Android).

kerel-fs, 2018-06-10:
Commit  b29143b (on my branch [fix/76](https://github.com/kerel-fs/XCSoar/tree/fix/76) ) fixes this issue, but I've no further insight yet whether this is the correct approach.

commit 7fb7829a (2018-08-27): MapWindow/Task: use dummy projection if OrderedTask is not active

Fixes assertion failure in `OrderedTask::GetTaskProjection()`.  Closes
#76.
closed 2018-08-27 by MaxKellermann