XCSoar/XCSoar #81: Build fails for target UNIX at v6.8.10.

Exadios, 2018-05-12:
Source file src/Screen/X11/TopWindow.cpp fails to build due to missing header.

Turbo87, 2018-05-13:
on what platform? seems to work fine for me 🤔

Exadios, 2018-05-18:
Hi Tobias;

I'm building on a standard issue Debian 9 ("stretch") using g++ v6.3.0 and its headers.

But rather than focusing on what might be different with our platforms I would like to make the following argument in favor of merging this pull.

I direct your attention to https://github.com/XCSoar/XCSoar/blob/v6.8.x/src/Screen/X11/TopWindow.cpp, line 60 - fprintf(...... So stdio.h is required. In your case it must be included as a result of a side effect. In my case it is not.

By leaving out the include here the source is dependent on the particular details of (presumably) the build environment. That is the omission of the include is a failure waiting to happen. On the other hand having the include here is quite natural in the circumstances and, if it redundant, it will be ifdefed out.

Exadios, 2018-05-19:
One other thing I should mention is that I run make with DEBUG=n and GREEN_MENU=n

kerel-fs, 2018-06-10:
Issue was fixed on the master in 2015 by Max, see https://github.com/XCSoar/XCSoar/commit/a09fa72bc0938ae0261c375415d83e6a0911f8f1

Please base all your development on the `master` branch. If you think something should be backported, rebase those exact commits.

Exadios, 2018-06-19:
So, it seems that I cannot reproduce the problem. I must have had some sort of finger problem here. This issue is closed.

closed 2018-06-19 by Exadios