XCSoar/XCSoar #119: Download of waypoint file `Germany.cup`via HTTPS fails on UNIX

kerel-fs, 2018-06-10:
Running current `master` / 4c6e754 on UNIX, downloading the file "Germany.cup" fails.

## Steps to reproduce
Start XCSoar, choose any mode (e.g. SIM), open the FileManager, Add "Germany.cup"

## Result
Germany.cup appears in the file list only for a really short time and then disappears again. Additionally the phrase `Out of memory` appears on stderr.

## Debug steps
Backtrace when stepping through the code:
```
(gdb) c
Continuing.
cannot open /home/kerel//.xcsoar/lua/init.lua: No such file or directory
[New Thread 0x7fffe980e700 (LWP 1479)]
[New Thread 0x7fffe3fff700 (LWP 1480)]
[Thread 0x7fffe3fff700 (LWP 1480) exited]
[New Thread 0x7fffe3fff700 (LWP 1496)]
[Thread 0x7fffe3fff700 (LWP 1496) exited]
[Switching to Thread 0x7fffe980e700 (LWP 1479)]

Thread 6 "DownloadMgr" hit Breakpoint 1, write () at ../sysdeps/unix/syscall-template.S:84
84	in ../sysdeps/unix/syscall-template.S
(gdb) bt
#0  write () at ../sysdeps/unix/syscall-template.S:84
#1  0x00007ffff5263817 in _IO_new_file_write (f=0x7ffff558b520 <_IO_2_1_stderr_>, data=0x7fffe980b680, n=14) at fileops.c:1271
#2  0x00007ffff5262b62 in new_do_write (fp=fp@entry=0x7ffff558b520 <_IO_2_1_stderr_>, data=data@entry=0x7fffe980b680 "Out of memory\n", 
    to_do=to_do@entry=14) at fileops.c:526
#3  0x00007ffff5263f6e in _IO_new_file_xsputn (f=0x7ffff558b520 <_IO_2_1_stderr_>, data=<optimized out>, n=14) at fileops.c:1350
#4  0x00007ffff523acc5 in buffered_vfprintf (s=0x7ffff558b520 <_IO_2_1_stderr_>, format=<optimized out>, args=<optimized out>)
    at vfprintf.c:2346
#5  0x00007ffff5237f25 in _IO_vfprintf_internal (s=0x7ffff558b520 <_IO_2_1_stderr_>, format=0x5555559fb97a "%s\n", 
    ap=ap@entry=0x7fffe980dc28) at vfprintf.c:1293
#6  0x00007ffff5240187 in __fprintf (stream=<optimized out>, format=<optimized out>) at fprintf.c:32
#7  0x0000555555680acc in LogString (p=0x7fffe4022078 "Out of memory") at src/LogFile.cpp:83
#8  0x0000555555680ccb in LogError (exception=...) at src/LogFile.cpp:152
#9  0x0000555555996045 in DownloadManagerThread::ProcessQueue (this=0x555555f21080, session=...) at src/Net/HTTP/DownloadManager.cpp:293
#10 0x000055555599505a in DownloadManagerThread::Tick (this=0x555555f21080) at src/Net/HTTP/DownloadManager.cpp:322
#11 0x000055555599e17e in StandbyThread::Run (this=0x555555f21080) at src/Thread/StandbyThread.cpp:115
#12 0x000055555599c8ae in Thread::ThreadProc (p=0x555555f21080) at src/Thread/Thread.cpp:166
#13 0x00007ffff603c494 in start_thread (arg=0x7fffe980e700) at pthread_create.c:333
#14 0x00007ffff52d9acf in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:97
```
So, the error is logged by this statement:
https://github.com/XCSoar/XCSoar/blob/master/src/Net/HTTP/DownloadManager.cpp#L293
Cause by an exception raised by this statement:
https://github.com/XCSoar/XCSoar/blob/fb18d530575a1126400554af65d3ee582d1c809a/src/Net/HTTP/ToFile.cpp#L95
inside this function
https://github.com/XCSoar/XCSoar/blob/fb18d530575a1126400554af65d3ee582d1c809a/src/Net/HTTP/Request.cpp#L100

## (preliminary) Explanation
Currently `Germany.cup` is the only waypoint file using HTTPS. The `libcurl` error code `CURLE_OUT_OF_MEMORY` is translated by `curl_easy_strerror` to the above mentioned error message.

A similar error was reported 2008 on the ML curl-library: http://curl-library.cool.haxx.narkive.com/RwCtIULD/https-request-with-curl-library-gives-out-of-memory-error

kerel-fs, 2018-07-07:
Similar error report & possible fix: https://stackoverflow.com/questions/166617/curle-out-of-memory-error-when-posting-via-https#5580270

kerel-fs, 2018-07-07:
Currently libcurl is built with the flag `--without-ssl` in xcsoar, see
https://github.com/XCSoar/XCSoar/blob/master/build/python/build/libs.py#L130

MaxKellermann, 2020-06-13:
I can't reproduce this problem, but the `--without-ssl` option is not used for the Linux build, because the Linux build uses the system CURL library. And in any case, this sounds very much like a problem in your system CURL build.

closed 2020-06-13 by MaxKellermann