XCSoar/XCSoar #120: XCSoar seems to crash on Android 8.0

lordfolken, 2018-06-12:
Multiple reports in the forum, that Android 8 breaks current stable on startup.

Currently investigating...

kicune, 2018-06-12:
Can confirm that, happens on my device too. Let me know if you need some logs or any other help.

lordfolken, 2018-06-12:
@kicune can you dump the adb logcat here in the ticket?

lordfolken, 2018-06-12:
i tested an android 8.1 device with both stable and master. No issues. Is there some aggressive powersaving that is causing the problem?

felixhaedicke, 2018-06-13:
Please check if commit fc80aaf (currently on 6.8.x branch) fixes this.

flypluswangzhe, 2018-06-19:
I have tanslate all word to Chinese , where I can download newest version , such as ver6.8.11 ,and I can test my tanslate word ,my email address is 216188@qq.com thanks,

kicune, 2018-06-19:
Sorry for the delay, I was travelling. Here is the logcat output.
[XCSoarLog.txt](https://github.com/XCSoar/XCSoar/files/2114682/XCSoarLog.txt)

SIGSYS arrived at 11:30:55.808. As it was related to "Terrain", I tried to switch off the terrain map; it dumped again, this time within "Topography".

For the record, this problem is caused only by version 7.0 compiled by me. Both 7.0 and 6.8 from the app store works, so it may be caused by something from my compile (I used vagrant env. from vanilla master branch).

kicune, 2018-06-19:
I backported fc80aaf to master (have some compilation problems with v6.8,x atm) and I can confirm that - as far as I was able to check - it works correctly!

lordfolken, 2018-06-22:
@kicune can you create a pull request?

kicune, 2018-06-23:
Here: https://github.com/XCSoar/XCSoar/pull/122
I am not sure about the proper way how to do this though, let me know if I should amend something.

MaxKellermann, 2018-08-27:
fc80aafbbf35eacd26aa9c089f7cf41680efd9ba

closed 2018-08-27 by MaxKellermann