XCSoar/XCSoar #123: Black terrain color ramp

georgebohnisch, 2018-07-02:
Using a device with an OLED panel, having a black terrain color ramp would greatly reduce battery consumption.

georgebohnisch, 2018-07-19:
To clarify on this, the base color would be black, so the majority of the terrain would be black (OLED displays do not expend power on pixels that are black/off) and the terrain would be lighter shades of gray/white.

MikeBravo15, 2018-08-15:
I can do that on my Nokia8 via colors inverse function.  Visibility  is OK but not as good as white as screen reflexion is pronounced. Issue is that sun heats the screen significantly so beware od that

lordfolken, 2020-05-04:
This would be fairly easy to do in:
https://github.com/XCSoar/XCSoar/blob/82bc641f038c8620ab534a8f203b8549bf26ff54/src/Terrain/TerrainRenderer.cpp

please contribute.

fberst, 2020-10-18:
i have implemented the black color ramp: https://github.com/fberst/xcsoar/commit/b6e8278360726a2bf7e5b13ea14bb18639438b74

lordfolken, 2020-10-18:
@fberst please create a pull request

lordfolken, 2021-07-02:
Closing as stale.

closed 2021-07-02 by lordfolken