XCSoar/XCSoar #134: Lock screen feature request

pqueiros, 2018-08-19:
I have been using xcsoar in a kobo which is strapped to my leg and keep having the problem of accidentally (probably with the risers or the harness) 'click' on the screen causing all sorts of misconfigurations.

I am not sure if others have this issue but I would imagine so.  This since I could not find any reference to a screen lock feature. I would like to suggest to implement one.

I am thinking an option under the display menu called "lock screen". Once button is selected only the main menu button would be selectable and once clicked would show a popup in some other of the screen to confirm whether the screen should be unlocked or not. Once unlock confirmation popup is shown any other click outside the confirmation button would cancel the action and go back to main menu.

Ideas and or suggestions are very welcome.
Regards,
pq

pqueiros, 2018-08-27:
Hi,

I started looking at the source code and to the many events that can be programmed but I am a bit overwhelmed :). this thing is really complex. does anyone have pointers how to get into the code base. And even better suggestions in how to implement this feature?

MaxKellermann, 2018-08-27:
You could implement this as a small modal dialog ([`class WndForm`](https://github.com/XCSoar/XCSoar/blob/master/src/Form/Form.hpp)) which is displayed in one screen corner. It contains just one small "Unlock" button. That way, XCSoar shows the usual information, but does not respond to clicks (only to the "Unlock" button).
Once we have that, we can refactor it to an invisible overlay which requires the user to draw a gesture (dimming the screen, showing the gesture and instructions while the finger touches the screen). But let's do one thing at a time.

MaxKellermann, 2018-08-27:
Oh and yes, one can easily be overwhelmed by the complexity of XCSoar. And its heavily underdocumented non-standard UI toolkit. I hope you have enough patience to get this done.

pqueiros, 2018-08-28:
@MaxKellermann,

Thank you very much for the hints. I will look for references on the WndForm class and investigate further.

ulipo, 2019-03-25:
Any news on this front? I think it would be a very useful feature for any praraglider pilot.
LJK8000 has a feature that disables touch screen until the software detects the take off, but locking the scrren after a time out would be definiteli more useful option.

nico-krais, 2019-05-31:
I started to implement a feature like that in https://github.com/nico-krais/XCSoar/tree/ScreenLock. As suggested, there is a new button under "Display" that locks the screen, and only shows a very, very small OK button to return to normal operation.
I would much prefer to have it that a longer press on the button is required instead of making it very small, but couldn't figure that out by now. 
Also, it would be very helpful to automatically unlock the screen at take-off - at least that would be my use-case.
If anyone with more experience in the code-base could point me in the right direction for the two points mentioned above, I would also have a look at that.

nico-krais, 2019-06-04:
I now tested this successfully on my Kobo mini, it prevents exactly what @pqueiros described: That the risers trigger all sort of Inputs before take off. Until now I have not managed to unlock the screen unintentionally. The new menu item looks like this:

[image]
and after pressing the button you see
[image]
The current version of my implementation can be found here: https://github.com/nico-krais/XCSoar/tree/v6.8.11_patched

I based this on the v6.8.11 tag since this is the current release and I have that in production. 

As mentioned above this is probably as far as I can go without some directions from a more experienced developer.

If this feature is of interest, I can prepare a merge request. Let me now if specific work should be done before.

If someone is interested in the meantime, I uploaded the version I use on my Kobo mini: https://github.com/nico-krais/XCSoar/releases/tag/v6.8.11_patched - WITHOUT ANY SORT OF WARRANTY, THIS IS BARELY TESTED, ASSUME THAT IT WILL DESTROY YOUR DEVICE AND AT THE VERY LEAST HAVE A FULL BACKUP OF YOUR SD CARD

nico-krais, 2019-06-05:
I've just seen that there already exists a pull request for that, so it should be mentioned in this issue: #202 

So I won't prepare the merge request for my implementation.

rewopen, 2019-06-09:
I prefer to lock the screen permanently, while I am flying with the paraglider. So it is impossible that my jacket push on the screen from my kobo. When I lock the screen, I can see the „Screenlock“ and „Anzeige 3/3“ Buttons. It would be nice, when only a very small Button to unock was visible. Thanks

 

 <https://user-images.githubusercontent.com/20615486/58908034-ccb23100-870f-11e9-89c9-04c2f17d2616.png> 
 <https://user-images.githubusercontent.com/20615486/58908048-d8055c80-870f-11e9-9ee3-a3b85d7c8580.png>

ulipo, 2019-06-09:
My implementation, the one in the (somehow embarrassing) pull request, uses a modal dialog like the one of Nico Krais, but I tailored it to be as small as possible and to put it in an area where it shouldn't cover anything important.
Lately I made a small change (the button acts now as a "cancel" instead of "OK") and that brought in the unexpected ability to close the modal also by pressing the hardware "power" button!

If you want to try it out, here attached is an updated KoboRoot.tgz, but use it at your own risk, I'm not responsible if it bricks your device, raids your fridge, kills your dog and so on...
[unzip_me.zip](https://github.com/XCSoar/XCSoar/files/3269524/unzip_me.zip)

nico-krais, 2019-06-09:
> I prefer to lock the screen permanently, while I am flying with the paraglider. So it is impossible that my jacket push on the screen from my kobo. When I lock the screen, I can see the „Screenlock“ [...]

Yea, the file I uploaded was from an old commit - it's updated now. Also the dialog is smaller. But again, since there is the pull request by @ulipo, the smartest thing is to wait for that to get merged and officially released.

rewopen, 2019-07-07:
If I lock my screen on kobo, there is no thermal asstistant on the left bottom corner. If I unlock the creen, the asstistant is visible in cycling mode. I testet it in replay and while I am flying.

ulipo, 2019-07-07:
I guess the thermal assistant is done as another dialog window or such, probably it's queued to the modal window of the lock screen and isn't displayed until the first one is closed.
The same happens for the messages, they are all displayed in a row when you unlock the screen.
I am not skilled enough to know how to circumvent this, I am afraid for the time being you'll have to choose to have either the screen locked, or the other functions...

rewopen, 2019-07-07:
Thank you. It is possible to use an event.xci? Do you know the syntax for thermal asstistant window?

ulipo, 2019-07-07:
I'm sorry, I totally don't know

rewopen, 2019-07-08:
I tried the thermalAsstist.xci
Switch automatically to Thermal Assist screen when thermalling, switch back to previous screen when not.
   Copy thermalAssist.xci to XCSoarData
    Start XCSoar goto:
    Config 2/3 -> System -> Look -> Language, Input
    Select Events then thermalAssist.xci
    Restart XCSoar
    
    mode=default
    type=gce
    data=FLIGHTMODE_CLIMB
    event=ThermalAssistant
    event=Mode default

    mode=default
    type=gce
    data=FLIGHTMODE_CRUISE
    event=ScreenModes previous
    event=Mode default
It works an make the ThermalAsstistant visible, also if the screen ist locked. But I would prefer the small ThermalAssistant in the left bottom corner. So I would see my flight track. The thermalAssit.xci I found from Lloyd on the internet.

ajdayvie, 2019-08-24:
Am I missing something obvious here. I've tried to update my build on kobomini using all of the KoboRoot.tgz files above. none of them show the menu item DISPLAY3/3 > Screen Lock. After updating my kobo shows a new version number so it seems like it's loading the files ok.
I would love to have this feature. I'm always having to reset stuff mid flight because all of the unintentional buttons pushed.

ulipo, 2019-08-24:
If you use my build from the "unzip_me.zip" file I posted above, the lock screen button is in the main menu, not in DISPLAY.

ulipo, 2019-08-26:
That's strange, the button should be just on the left of "Cancel" in main menu, I already tried it on a Mini and I'm sure it was working...
Try this other build here, it's the one I'm using on my Kobo Glo and includes also the OTG kernel (but I didn't test that on Mini)
[KoboRoot_XCsoar_OTG_LOCK.zip](https://github.com/XCSoar/XCSoar/files/3542555/KoboRoot_XCsoar_OTG_LOCK.zip)
As always, take a full backup of your SD card before trying stuff, or anyway don't hold me responsible if your Kobo blows up in chippings :grin:

lordfolken, 2020-05-08:
This is implemented in preview15

closed 2020-05-08 by lordfolken