XCSoar/XCSoar #137: make error

hatietz, 2018-08-25:
Hello,
I got this error:
tietz@tietz-EX58-UD5:~/XCSoar$ make TARGET=UNIX64
  UNTAR   boost_1_65_1.tar.bz2
/bin/sh: 1: quilt: not found
build/libboost.mk:18: recipe for target 'output/src/stamp-boost_1_65_1' failed
make: *** [output/src/stamp-boost_1_65_1] Error 127

My environment:
tietz@tietz-EX58-UD5:/etc$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 16.04.5 LTS
Release:	16.04
Codename:	xenial
tietz@tietz-EX58-UD5:/etc$ uname -i
x86_64

I used this manual to install the development:
https://download.xcsoar.org/releases/6.8.11/XCSoar-developer-manual.pdf

This shows in short form the steps to provide and install the development environment:
sudo apt-get clean
sudo apt-get update
sudo apt-get upgrade

sudo apt-get install make \
librsvg2-bin xsltproc \
imagemagick gettext \
git

sudo apt-get install make g++ \
zlib1g-dev \
libsdl1.2-dev libfreetype6-dev \
libpng-dev libjpeg-dev \
libcurl4-openssl-dev \
libxml-parser-perl \
librsvg2-bin xsltproc \
imagemagick gettext \
libegl1-mesa-dev \
libgl1-mesa-dev \
fonts-dejavu

git clone git://github.com/XCSoar/XCSoar
cd XCSoar
git pull
git submodule init
git submodule update
make TARGET=UNIX64

hatietz, 2018-08-25:
After sudo apt-get install quilt and 
make TARGET=UNIX64
the error was gone

closed 2018-08-25 by hatietz

hatietz, 2018-08-25:
I did a apt-get install quilt --> the was gone but i got 
/usr/lib/gcc/x86_64-linux-gnu/5/include/xmmintrin.h:72:1: internal compiler error: Segmentation fault
 typedef float __v4sf __attribute__ ((__vector_size__ (16)));
 ^
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-5/README.Bugs> for instructions.
build/compile.mk:111: recipe for target 'output/UNIX64/dbg/src/Renderer/UnitSymbolRenderer.o' failed

I did "make TARGET=UNIX64" again and that error was gone but I got this
Package lua5.2 was not found in the pkg-config search path.
Perhaps you should add the directory containing `lua5.2.pc'
to the PKG_CONFIG_PATH environment variable
No package 'lua5.2' found
build/compile.mk:111: *** library not found: lua5.2.  Stop.

I installed --> sudo apt-get install lua5.2 but the error was still there so i installed
sudo apt-get install liblua5.2-dev
and after it i could see 
ietz@tietz-EX58-UD5:~/XCSoar$ ls /usr/lib/x86_64-linux-gnu/pkgconfig/lua5*
/usr/lib/x86_64-linux-gnu/pkgconfig/lua5.2-c++.pc
/usr/lib/x86_64-linux-gnu/pkgconfig/lua52-c++.pc
/usr/lib/x86_64-linux-gnu/pkgconfig/lua5.2.pc
/usr/lib/x86_64-linux-gnu/pkgconfig/lua52.pc

after make TARGET=UNIX64 i got this:
Package libtiff-4 was not found in the pkg-config search path.
Perhaps you should add the directory containing `libtiff-4.pc'
to the PKG_CONFIG_PATH environment variable
No package 'libtiff-4' found
build/compile.mk:111: *** library not found: libtiff-4.  Stop.

but libtiff-4 can not be installed on Ubuntu 16.04 so i Instaled ..-5
sudo apt-get install libtiff5-dev --> and got this error
src/Screen/Custom/LibTiff.cpp:36:21: fatal error: geotiff.h: No such file or directory
compilation terminated.
build/compile.mk:111: recipe for target 'output/UNIX64/dbg/src/Screen/Custom/LibTiff.o' failed
make: *** [output/UNIX64/dbg/src/Screen/Custom/LibTiff.o] Error 1

sudo apt-get install libgeotiff-dev
and again 
make TARGET=UNIX64 and got 
FFMPEG    output/data/sound/fail.raw
make: ffmpeg: Command not found
build/resource.mk:155: recipe for target 'output/data/sound/fail.raw' failed
make: *** [output/data/sound/fail.raw] Error 127

see https://tecadmin.net/install-ffmpeg-on-linux/
sudo add-apt-repository ppa:jonathonf/ffmpeg-3
sudo apt-get update
sudo apt-get install ffmpeg libav-tools x264 x265

and again 
make TARGET=UNIX64 and got NO error!!!!!!!!!!!!!!!!!!!!!!!!
to call --> tietz@tietz-EX58-UD5:~/XCSoar/output/UNIX64/bin$ ./xcsoar

Now, how to get the terrain map displayed:
double click in the area where the terrain map should be
click on config until Config 1/3 shows up
click System
click Map Display
click Terrain  --> set Terrain DisplayOn, set Topo.. display On, set Terr.. Low lands
click close
click close

double click in the area where the terrain map should be 
click on config until Config 2/3 shows up
click FileManager
add
select  GER_HighRes.xcm
click Download
click close

double click in the area where the terrain map should be
click on config until Config 1/3 shows up
click System
click Site Files 
again click Site Files
click in "Map Database"
Highlight the map you want and click Select and than click close and again close
Now it is loading the Terrain File --> what a job to get is done!

MaxKellermann, 2018-08-26:
This bug report confuses me. Clearly "internal compiler error: Segmentation fault" is a bug in the compiler you installed, but this project can't fix your compiler.

hatietz, 2018-08-27:
That is right related to seg.. fault.
As i wrote -->  I just repeated the "make UNIX64" and the error (seg..
fault) was gone.
But I had to do a lot of addit'l installs not mentioned in the
XCSoar-developer-manual.pdf

sbrodehl, 2019-06-25:
I think `quilt` is still missing in [XCSoar-developer-manual.pdf](http://download.xcsoar.org/releases/6.8.11/XCSoar-developer-manual.pdf), how is this PDF generated?

sbrodehl, 2019-06-25:
Just created #224 .