XCSoar/XCSoar #139: Possible null pointer dereference

Exadios, 2018-09-03:
In src/Form/DataField/Prefix.cpp, PrefixDataField::GetAsDisplayString() it is possible that 's' is null.
    const TCHAR *s = DataFieldString::GetAsDisplayString();
    if (StringIsEmpty(s))

MaxKellermann, 2018-09-03:
How can this happen?

MaxKellermann, 2018-09-13:
Closing due to lack of feedback.

closed 2018-09-13 by MaxKellermann

Exadios, 2018-09-17:
I thought that this issue had been closed some time ago.

What feedback where you expecting?

MaxKellermann, 2018-09-17:
#140 was closed two weeks ago, because it was the wrong approach to fix this bug report.
I was expecting feedback as in "answering my question".

Exadios, 2018-09-21:
OK. But I will have to go over it again.

WRT XCSoar I'm presently working on a program termination and (what I am supposing to be) associated valgrind so I'll continue with that.