XCSoar/XCSoar #147: Wifi not working on Kobo Clara HD

burakcan, 2018-09-29:
Hello,
I've installed XCSoar to Kobo Clara HD. Almost everything works but when I go to "Network" and tap to "WIFI ON"; nothing happens and "Wifi" button stays inactive.
XCSoar version: 6.8.11

burakcan, 2018-09-30:
So.. I've built latest master (67a9a193fa319029dc95b4f4c22c5e9d38a9884f) and ran ./KoboMenu. Wifi works now. It's either fixed somewhere in between or starting Nickel before XCSoar changes this behavior, not sure yet. 
XCSoar's rcS is booting but the screen is not responsive and activity led keeps blinking. I'll try to boot to XCSoar with an edited rcS.

burakcan, 2018-10-04:
I've tried it and confirmed that XCSoar cannot turn wifi on or off but can use the network if it's connected before launching XCSoar.

burakcan, 2018-10-08:
It didn't work. @MaxKellermann do you have any hints that you can give me to track this?

lordfolken, 2018-10-12:
You are right in that this device is not a target at the moment. So your mileage may vary, and this is such a case. If you have a look at WifiDialog.cpp  Line 320 there is an expected device path that it looks for on the system:     
wpa_supplicant.Connect("/var/run/wpa_supplicant/eth0");

Does that exist on your device?

burakcan, 2018-10-12:
@lordfolken thank you for the comment. I think it looks for eth0 file after turning the wifi on. In Clara HD case, turning wifi doesn't work yet. It happens here: 
https://github.com/XCSoar/XCSoar/blob/719cedbf614b037c75081fa86f7be3735fb0daf4/src/Kobo/System.cpp#L175

I checked those driver files while nickel is running and the drivers that are used for AURA2 case seems to be there. 
```
    InsMod("/drivers/mx6sl-ntx/wifi/sdio_wifi_pwr.ko");
    InsMod("/drivers/mx6sl-ntx/wifi/8189fs.ko");
```
To try, I removed the switch/case there and left only those InsMod calls and compiled. But it still can not turn wifi on. Not sure what the problem is.

lordfolken, 2018-10-12:
I see later in the code that ifconfig eth0 up is called. ifconfig is part of a toolsuite that is considered deprecated by many linux distributions. THe new tool is iproute2 its command is "ip". Could you try to execute ifconfig eth0 down from a terminal into device and see if it finds the command at all?

burakcan, 2018-10-12:
ifconfig is there and seems to be working but iwconfig (`Run("/sbin/iwconfig", "eth0", "power", "off");`) says "operation not permitted". (trying through telnet over usb network, while nickel is running)

pyctam, 2019-07-31:
> I solved it by switching to Aura 2

1) Do you mean that I had you build custom version which recognizes Clara HD as Aura 2?
2) If so, was your fix rolled into 6.8.11?

burakcan, 2019-08-01:
@pyctam no, I bought an Aura2 and using Clara HD just for reading :)

wolkenjaeger, 2021-04-18:
I try to revive the discussion regarding Kobo Clara HD again. I also made two suggestions in the Code to add the model. However, my coding skills are not as good to be an active coder or developer here.