XCSoar/XCSoar #157: compile out of the box: master (commit b45fe171bc11b6eabee474bc7978d98c786f147a)

ThHanika, 2018-12-22:
Hi, Since I wanted to dedicate myself to my "pull request" again and want to test the usability of my patch, I have cloned pure from the master repository xcsoar and compiled. Unfortunately
With the current master version no working xcsoar for Linux (Debian) can be produced. The presentation is not OK.
commit b45fe171bc11b6eabee474bc7978d98c786f147a

Turbo87, 2018-12-22:
can you use `git bisect` to check what commit introduced these issues for you?

ThHanika, 2018-12-22:
OK this is the bad one:
67a9a193fa319029dc95b4f4c22c5e9d38a9884f
[image]

this is the last good one:
faf7e212cc3e736cba25b98107098069d3eaf3e9
[image]

Turbo87, 2018-12-22:
sounds like your system does not support GLSL 🤔

ThHanika, 2018-12-22:
It seems YES:
OpenGL vendor string: Intel Open Source Technology Center
OpenGL renderer string: Mesa DRI Intel(R) Sandybridge Mobile 
OpenGL core profile version string: 3.3 (Core Profile) Mesa 13.0.6
OpenGL core profile shading language version string: 3.30
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile
OpenGL core profile extensions:
OpenGL version string: 3.0 Mesa 13.0.6
OpenGL shading language version string: 1.30
OpenGL context flags: (none)
OpenGL extensions:
OpenGL ES profile version string: OpenGL ES 3.0 Mesa 13.0.6
**OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.00**
OpenGL ES profile extensions:

Turbo87, 2018-12-23:
interesting, then I'm out of ideas. maybe @MaxKellermann know more about this.

rawtaz, 2019-01-29:
@ThHanika What command should one run to check the GL stuff like you did above? It never hurts to include commands with the corresponding output ;)

ThHanika, 2019-02-05:
This commandline:

glxinfo | grep "OpenGL"

ThHanika, 2019-02-05:
**thomas@Dell-Thomas:~$ glxinfo | grep "OpenGL"**
OpenGL vendor string: Intel Open Source Technology Center
OpenGL renderer string: Mesa DRI Intel(R) Sandybridge Mobile 
OpenGL core profile version string: 3.3 (Core Profile) Mesa 13.0.6
OpenGL core profile shading language version string: 3.30
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile
OpenGL core profile extensions:
OpenGL version string: 3.0 Mesa 13.0.6
OpenGL shading language version string: 1.30
OpenGL context flags: (none)
OpenGL extensions:
OpenGL ES profile version string: OpenGL ES 3.0 Mesa 13.0.6
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.00
OpenGL ES profile extensions:
thomas@Dell-Thomas:~$

ThHanika, 2019-02-11:
Is it working on other OpenGL Version?

MaxKellermann, 2019-02-11:
I have no such problem with the amdgpu driver and a Radeon RX 480.

ThHanika, 2019-02-11:
Please check your OpenGL Version with following command and make it public:

glxinfo | grep "OpenGL"

MaxKellermann, 2019-02-11:
```
OpenGL vendor string: X.Org
OpenGL renderer string: AMD Radeon (TM) RX 480 Graphics (POLARIS10, DRM 3.27.0, 4.20.6-heron, LLVM 7.0.1)
OpenGL core profile version string: 4.5 (Core Profile) Mesa 18.3.2
OpenGL core profile shading language version string: 4.50
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile
OpenGL core profile extensions:
OpenGL version string: 4.5 (Compatibility Profile) Mesa 18.3.2
OpenGL shading language version string: 4.50
OpenGL context flags: (none)
OpenGL profile mask: compatibility profile
OpenGL extensions:
OpenGL ES profile version string: OpenGL ES 3.2 Mesa 18.3.2
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20
OpenGL ES profile extensions:
```

ThHanika, 2019-02-12:
OK, it is not the INTEL-driver... 

Version of OpenGL ES GLSL ES 3.xx ?
Which version is used in Android, Openvario and Linux?

OpenGL vendor string: VMware, Inc.
OpenGL renderer string: Gallium 0.4 on llvmpipe (LLVM 3.9, 256 bits)
OpenGL core profile version string: 3.3 (Core Profile) Mesa 13.0.6
OpenGL core profile shading language version string: 3.30
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile
OpenGL core profile extensions:
OpenGL version string: 3.0 Mesa 13.0.6
OpenGL shading language version string: 1.30
OpenGL context flags: (none)
OpenGL extensions:
OpenGL ES profile version string: OpenGL ES 3.0 Mesa 13.0.6
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.00
OpenGL ES profile extensions:

ThHanika, 2019-02-12:
OK I installed **ubuntu-18.04.1-desktop-amd64** 

OpenGL vendor string: VMware, Inc.
OpenGL renderer string: llvmpipe (LLVM 7.0, 256 bits)
OpenGL core profile version string: 3.3 (Core Profile) Mesa 18.2.2
OpenGL core profile shading language version string: 3.30
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile
OpenGL core profile extensions:
OpenGL version string: 3.1 Mesa 18.2.2
OpenGL shading language version string: 1.40
OpenGL context flags: (none)
OpenGL extensions:
OpenGL ES profile version string: OpenGL ES 3.0 Mesa 18.2.2
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.00
OpenGL ES profile extensions:

### **this works!!!!!**

ThHanika, 2019-02-12:
It seems, it is a mesa version problem

ThHanika, 2019-02-13:
Bin auf XUbuntu gewechselt: Version 18.04.1 = geht!!!

OpenGL vendor string: Intel Open Source Technology Center
OpenGL renderer string: Mesa DRI Intel(R) Sandybridge Mobile 
OpenGL core profile version string: 3.3 (Core Profile) Mesa 18.2.2
OpenGL core profile shading language version string: 3.30
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile
OpenGL core profile extensions:
OpenGL version string: 3.0 Mesa 18.2.2
OpenGL shading language version string: 1.30
OpenGL context flags: (none)
OpenGL extensions:
OpenGL ES profile version string: OpenGL ES 3.0 Mesa 18.2.2
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.00
OpenGL ES profile extensions:

ThHanika, 2019-02-13:
OK Closed, it is not a xcsoar development problem.

closed 2019-02-13 by ThHanika