XCSoar/XCSoar #163: XCSoar crash on Linux

mkauzlar, 2019-01-20:
Running latest git clone on linux.
When connecting Condor2 over TCP 4353 all works well and it connects. However after approximately 30sec of moving (the glider) it crashes.
The output is:
xcsoar: src/IGC/IGCWriter.cpp:107: void IGCWriter::WriteHeader(const BrokenDateTime&, const TCHAR*, const TCHAR*, const TCHAR*, const TCHAR*, const char*, const TCHAR*, bool): Assertion `date_time.IsPlausible()' failed.
Aborted (core dumped)

Exadios, 2019-01-21:
So, to be clear, you have cloned XCSoar/XCSoar, then compiled the master branch with "DEBUG=Y" and got the result you have described. Is that correct.

Could you turn on 'NMEA logging', repeat and then replay the recorded NMEA file to see if you get the same failure. If so, post that NMEA file here.

mkauzlar, 2019-01-21:
Yes, I cloned and  compiled the master branch  however I did not set DEBUG=Y (I don't know whether that's the default setting). The result is printed out on the terminal when crashed.
I'm building from source because the deb packages don't work on Ubuntu 18.04.

Here is the procedure for building on Ubuntu 18.04:
1. Install dependencies:
apt install quilt libclalsadrv-dev libcurl4-openssl-dev libglu1-mesa-dev liblua5.2-dev zlib1g-dev libfreetype6-dev libjpeg-turbo8-dev libtiff5-dev libgeotiff-dev xsltproc librsvg2-bin libglm-dev ffmpeg
2. Clone
git clone https://github.com/XCSoar/XCSoar.git
3. Build
cd XCSoar
make
4. Run
./output/UNIX/bin/xcsoar

I will report the result with NMEA logging once I have it

Exadios, 2019-01-21:
Yes, I've just checked, DEBUG = y is the default.

BTW you can check the value of this make variable by:
make -pn | grep DEBUG\ *=

mkauzlar, 2019-01-27:
the igc file is created but it's empty. The last log in the xcsoar is:

[2019-01-27T16:28:56Z] Register serial devices
[2019-01-27T16:28:56Z] Opening device TCP port 4353
[2019-01-27T16:28:56Z] ProgramStarted
[2019-01-27T16:28:56Z] Free storage 405130696 KB
[2019-01-27T16:28:56Z] cannot open /home/user//.xcsoar/lua/init.lua: No such file or dir
ectory
[2019-01-27T16:35:47Z] Using ALSA PCM device "default" (use environment variable ALSA_DEVICE t
o override)
[2019-01-27T16:35:47Z] Using ALSA PCM latency 100000 μs (use environment variable ALSA_LATENCY
 to override)
[2019-01-27T16:36:42Z] ALSA PCM buffer underrun
[2019-01-27T16:36:42Z] ALSA PCM successfully recovered
[2019-01-27T16:36:42Z] Logger Started: /home/user//.xcsoar/logs/2019-01-27-XCS-BSA-02.ig
c

Exadios, 2019-01-28:
So, I think we have a bit of a communication problem. The IGC file is of no interest.

I want you to turn on NMEA logging (Config->System->Setup->Logger->NMEA Logger->On) and then run your Condor2 as you were. This should record a file with a date / time as a filename and 'nmea' as the extension. This should contain the NMEA records sent by Condor2 to XCSoar. Then I want you replay this file (Config->Config->Replay) using XCSoar to confirm that replaying this file triggers the problem previously noted.

If this file does trigger the problem we then have the NMEA records causing the fault. Then it is a relatively easy task for me (or anybody else) to replay the file into XCSoar running under the debugger to diagnose and fix the problem. By fixing the problem triggered by the file the problem with Condor2 will be fixed as well.

There is a chance that this method will not work because the NMEA will not be flushed before the assert(). If this is the case then plan B is to use ncat which, I'm guessing, you do not know how to use and, so, will be more difficult.

MaxKellermann, 2019-01-28:
This means XCSoar has crashed before it could flush the file.
netcat: try `nc 11.22.33.44:4353 |tee /tmp/nmea` (replace `11.22.33.44` with the IP address of the Condor computer; no XCSoar)
Wait some time; you should be seeing NMEA on your terminal, and additionally in the `/tmp/nmea` file. Press Ctrl-C to stop netcat. Upload that file.

mkauzlar, 2019-01-30:
this nc comand doesn't run on linux because of wrong format and because the Windows machine with Condor2 is not listening to 4353.
If the goal is to capture the nmea on the Linux machine I did that with: 
nc -l -p 4353 | tee condor2_nc_nmea
Which makes it listen to port 4353 and capture the communication. Attached is the file.
I replayed this nmea file in Condor without a problem.
[condor2_nc_nmea.zip](https://github.com/XCSoar/XCSoar/files/2810812/condor2_nc_nmea.zip)

Exadios, 2019-01-30:
Excellent. I can see that you have recorded over a minute of data - long enough to trigger the problem?

So, its weird that this file does not trigger the problem. Most of the logic is the same whether the input is via the port or via the replay mechanism. There is a 'nmeacat' around somewhere. I'll dig it up and playback through a port. But I think this is a long shot.

Can you post the git hash of the XCSoar that you have compiled - find this from git or from within XCSoar by
INFO->INFO->CREDITS.

I wont be able to do any significant work on this until I go down to the club again - that may be a few weeks. Somebody else maybe able to get to this faster. In the meantime you may be able to continue with Condor by using a production release in which the abort()s are disabled. To make this proceed as you outlined above but use

make DEBUG=n TARGET=UNIX

mkauzlar, 2019-02-01:
I compiled also the 6.8 version and the result is the same (on 2 PCs)
I have now made a short flight and recorded the nmea with netcat:

[condor2_nc_nmea.zip](https://github.com/XCSoar/XCSoar/files/2822003/condor2_nc_nmea.zip)

1. If I replay the nmea file within XCSoar it works (both 6.8 and 7).
2. If I feed the file with netcat or flight with the Condor simulator I get the crash. 

To stream the file I use:

`cat condor2_nc_nmea | awk  '{print $0 ; system("sleep 0.1")}' | nc -q1 localhost 4353`

MaxKellermann, 2019-02-02:
c21a23472be4baf773fa18f1cd900c0a39928d02

commit c21a2347 (2019-02-01): IGC/Writer: log date only if it is available

Closes #163 (assertion failure crash because Condor doesn't provide a
date).
closed 2019-02-02 by MaxKellermann