XCSoar/XCSoar #168: Startup profile selection menu

NotYourBest, 2019-02-19:
I have an OpenVario running XCSoar 7.0 preview, set-up to portrait mode, however the profile selection menu on startup initializes to landscape. After selection it switches to portrait orientation; this also happens when "Default" is selected. 
The problem seems to occur irregardless of the device screen orientation setting and the XCSoar setting.

MaxKellermann, 2019-02-19:
The orientation is a setting of your profile, but before you have selected a profile, no settings are available.

ThHanika, 2019-02-19:
You could try, you can do that with start-parameter of xcsoar: (-480x800 is the suitable parameter for openvario 7 "display). However I do not know which impact that has, if a touchdisplay is used. In any case, the display orientation must be reset in openvario.prf with this start parameter.

ThHanika, 2019-02-19:
OK, an additional startup parameter is needed and developed.

NotYourBest, 2019-02-20:
> The orientation is a setting of your profile, but before you have selected a profile, no settings are available.

That I understand, but is it possible to have the device select the system default orientation for this start-up screen? On a phone or tablet this is easy to overcome by manually rotating the device, but is not so much true for buit-in applications.

Just checked on android with v6.8 and v7, where it follows the device rotation properly.. I'll have a look at the OpenVario settings

MaxKellermann, 2019-02-20:
XCSoar does already use the system default. But is the system default really the one you wish to have? I don't know anything about OpenVario, I've never seen one. How does one specify the "system default" on an OpenVario?

NotYourBest, 2019-02-21:
The OpenVario sets the uboot framebuffer to rotate to the selected setting. It looks like that is the system default after system startup. (shell is also in that orientation from start-up) Changing the XCSoar start parameter, as ThHanika mentioned, doesn't play nice.

The strange thing is that all other XCSoar screens are rotated correctly when orientation "default" is selected.

kedder, 2020-02-23:
This might be related to regression described in #351.

MaxKellermann, 2020-06-13:
Is this fixed now?

lordfolken, 2021-04-06:
@NotYourBest can you please check with current release?

NotYourBest, 2021-04-15:
Can confirm it has been fixed.

lordfolken, 2021-04-15:
yeah! /close

closed 2021-04-15 by lordfolken