XCSoar/XCSoar #174: Integrate Satellite images and Rain Radar from DWD

dspreitz, 2019-03-21:
In 2018 the German Goverment updated the [DWD law](https://www.gesetze-im-internet.de/dwdg/__6.html) and mandated that certain DWD products are provided under an open data license.
The license under which the data is published can be found [here](https://www.dwd.de/DE/service/copyright/copyright_node.html).

The data itself is accessible via a a [geoServer](https://docs.geoserver.org) in various formats (GeoTiff, OpenLayer, ....) [here](https://maps.dwd.de/geoserver/web/wicket/bookmarkable/org.geoserver.web.demo.MapPreviewPage?1).

This enhancement proposed to enable XCSoar to access this data and provide it on request as screen overlay.

lordfolken, 2021-04-08:
IMO this should be doable. We have support for overlays in various formats and should "only" be an enhancement. Care to provide a PR?

dspreitz, 2026-08-28:
Picking this up after six years, because the situation has become considerably
more concrete — and because the projection work needed for it now exists.

## Two open routes, both without registration

**1. The DWD's own WMS.** `https://maps.dwd.de/geoserver/dwd/wms` declares
`Fees: none` and `AccessConstraints: https://www.dwd.de/copyright`, and offers
`EPSG:4326` and `EPSG:3857`. The radar layers, with their advertised time
dimensions:

| layer | what it is | cadence |
|---|---|---|
| `dwd:RADOLAN-RY` | quality-checked radar data | `PT5M` |
| `dwd:RADOLAN-RW` | gauge-adjusted composite | `PT10M` |
| `dwd:Niederschlagsradar` | precipitation radar | `PT5M` |
| **`dwd:Radar_rv_product_1x1km_ger`** | **RADVOR analysis *and forecast*** | `PT5M` |
| `dwd:Radar_wn-analysis_1x1km_ger` | German radar composite | `PT5M` |

The RADVOR layer is the interesting one: its time dimension extends **about two
hours into the future** — when I read the capabilities at 20:45 UTC it ran to
22:40. That is a nowcast, which no static radar image offers.

**2. The raw RADOLAN products on `opendata.dwd.de`.**
`/weather/radar/radolan/` holds `ry/`, `rw/`, `sf/`, `yw/`, each with a
`latest` alias so no timestamp guessing is needed. Sizes matter here:

| product | content | cadence | size |
|---|---|---|---|
| **RY** | quality-checked precipitation rate | 5 min | **75 kB** (bz2) |
| RW | gauge-adjusted hourly composite | 10 min | 75 kB |
| PG (`composite/pg/`) | classic radar picture, BUFR | 5 min | 44 kB |
| WN, RV (`composite/`) | 1×1 km composite / forecast | 5 min | ~7 MB each |

75 kB every five minutes for the whole of Germany is comfortably within reach
of a mobile connection. The 7 MB tar products are not.

The server README states it plainly: *"DWD offers weather and climate data free
of charge […] Access is granted without registration."*

## The trade-off between them

**WMS**: request a bounding box in `EPSG:4326`, get back a north-up image, so
the four-corner georeference `MapOverlayBitmap` uses is correct by
construction. No projection work at all. The cost is one request per map view,
which needs a cache and a request policy — the design proposed in #2997 for
satellite imagery would carry over unchanged, since both are WMS with a time
dimension.

**opendata RADOLAN**: one 75 kB file every five minutes covering all of
Germany, regardless of where the map is looking. Far cheaper on bandwidth,
works offline once fetched, and needs no per-view request logic at all. The
cost is a parser for the RADOLAN binary format, and a georeference that is
*not* a north-up lat/lon raster.

## Why the second route is cheaper than it looks

RADOLAN is published on a polar stereographic grid: standard parallel 60°N,
central meridian 10°E, spherical earth with R = 6370.04 km. #2966 implements
exactly that projection, because the DWD satellite images turn out to use the
same grid.

So the georeference for route 2 is already written and unit-tested. What it
additionally needs is the projection-accurate mesh described in #2997 — the
four-corner interpolation in `MapOverlayBitmap` is not accurate enough for this
projection, and `SliceQuadrilateral()` refines the rasterisation while
inheriting the corner error. That change is small: have `Draw()` ask the
overlay for its cell corners instead of interpolating them.

I would suggest starting with the WMS route, since it needs no renderer change
at all and gets the nowcast for free, and treating the opendata route as the
bandwidth-efficient follow-up for people flying on poor connections.

## Licensing

Both are covered by DWDG §6 and the DWD copyright page linked in the original
post; attribution is required, exact wording per that page. Worth keeping the
radar attribution separate from the EUMETSAT one, since a map may end up
showing both.

## Relation to the other issues

#2997 proposes satellite imagery as a map overlay from EUMETView, with a
request policy and cache design that this would share. #2966 adds the DWD
projection and is what makes route 2 viable. This issue is the radar half, and
it is the stronger of the two for a German glider pilot, because the RADVOR
nowcast has no equivalent anywhere else.

@lordfolken — you asked for a PR back in 2021. Happy to prepare one now that
the projection is settled; a word on which of the two routes you would prefer
to review would help before I start.

dspreitz, 2026-08-28:
@kobedegeest I guess this also ties into your WIP on image overlays, which @lordfolken mentioned in https://github.com/XCSoar/XCSoar/issues/2997#issuecomment-5457482758

dspreitz, 2026-08-28:
One correction to the comment above, which I should have qualified: **the open DWD radar covers Germany, not Europe.**

I measured it rather than assumed it. Sampling `dwd:Niederschlagsradar` across the continent and counting pixels in the uncompressed RGBA GeoTIFF, so the no-data fill can be told apart from real echo:

* Germany 10.1 % non-transparent across 10 colours — real echo.
* France 30.6 % non-transparent across **one** colour, `7e7e7e4d`, the no-data/no-echo mask: inside the grid, outside radar coverage.
* Spain, southern Italy, Norway, Ireland: **no pixels at all**.

The footprint is exactly the RADOLAN national grid, roughly 900 × 900 km. A box spanning 35–60°N and 10°W–30°E came back 9.4 % covered, against 9.7 % predicted for that grid by area.

So of the six radar areas pc_met offers, the open layers replace `rx`, `rxn`, `rxm` and `rxs` fully, `fa` (Alps) only along its northern edge, and `eu` (Europe) not at all. The pc_met European product is presumably rendered from the EUMETNET OPERA composite, which the DWD does not republish openly.

`dwd:Radar_eucom_zeros` invites a wrong guess by its name: it is titled "No Data No Echo Maske (global)", a mask layer rather than European imagery.

#2998 implements the German case and now documents the limit in the manual.

dspreitz, 2026-08-28:
Findings from working this through, including two corrections to what I wrote above.

## The source: EUMETNET OPERA, pan-European, CC BY 4.0

`https://s3.waw3-1.cloudferro.com/openradar-24h/YYYY/MM/DD/OPERA/COMP/` serves the European composite anonymously, no account. The API declares CC BY 4.0 in its `collections` endpoint. Extent per the API: **22.6°W–29.8°E, 35.9°N–70.6°N**, 19 contributing countries.

Products: `DBZH` (maximum reflectivity, **5 min**), `RATE` (instantaneous rain rate, 15 min), `ACRR` (hourly accumulation, 15 min). The filename pattern is deterministic — `OPERA@YYYYMMDDTHHMM@0@DBZH.tiff` — so no API call is needed to find the current frame.

**It is a proper Cloud-Optimized GeoTIFF**, which is what makes it usable on a mobile connection:

| IFD | size | resolution | tiles |
| --- | --- | --- | --- |
| 0 | 3800 × 4400 | 1 km | 72 |
| 1 | 1900 × 2200 | 2 km | 20 |
| 2 | 950 × 1100 | 4 km | 6 |
| 3 | 475 × 550 | 8 km | 2 |
| 4 | 237 × 275 | 16 km | **1** |

512 × 512 tiles, Deflate, `Accept-Ranges: bytes` confirmed. A whole frame is 4.1 MB, but a single tile is **5 kB median** (min 2.1 kB, max 194 kB) and covers 512 × 512 km. A Europe-wide overview costs one tile. Projection is Lambert Azimuthal Equal Area, centre 55°N/10°E, false easting 1 950 000 m, false northing −2 100 000 m — closed-form, roughly the same effort as the polar stereographic already written in #2966.

## Correction 1: the DWD's European image is *not* OPERA

I assumed it was, because the pc_met product is titled "Precipitation Intensity (EuCom)". It is not. Per the DWD's own aviation brochure, EuCom is the European extension of the DWD's **WN composite**, assembled from data exchanged with neighbouring services "aufgrund bilateraler Abkommen" — OPERA is not mentioned. Same region, largely the same source radars, different processing chain.

## Correction 2: the open DWD radar covers Germany, not Europe

Measured, not assumed: sampling `dwd:Niederschlagsradar` across the continent and counting pixels in the uncompressed RGBA GeoTIFF, France returns exactly **one** colour — `7e7e7e4d`, the no-data mask — while Spain, southern Italy, Norway and Ireland return no pixels at all. The footprint is the RADOLAN national grid: a box spanning 35–60°N and 10°W–30°E came back 9.4 % covered against 9.7 % predicted for a 900 × 900 km grid.

So the open DWD layers replace the German pc_met radar areas, but not `eu`, and only the northern edge of `fa`.

## Making OPERA look like the DWD product

The DWD colour ramp is obtainable exactly: `maps.dwd.de` serves its SLD style over `GetStyles`, 15 classes with `type="intervals"`. Verified against the pc_met image itself, the values agree to within one bit per channel.

Converting `DBZH` to a rate to drive that ramp turned out to be the interesting part. Comparing the two products pixel-by-pixel in the same frame at the same minute:

* Class medians give **Z = 326 · R^1.42**. The exponent matches the DWD's documented **Z = 256 · R^1.42** ([Radarniederschlag, DWD 2015](https://www.dwd.de/DE/leistungen/radarniederschlag/rn_info/download_niederschlagsbestimmung.pdf?__blob=publicationFile&v=4)) essentially exactly; the coefficient is 1.05 dB higher.
* But no single Z-R relation fits the whole range. Comparing OPERA `DBZH` directly against the DWD's own **WN composite** from `opendata.dwd.de` over Germany — 144 390 pixel pairs, both 1 km, same minute — the offset is **intensity-dependent**:

| WN reflectivity | median OPERA − WN |
| --- | --- |
| 5–15 dBZ | **+8.35 dB** |
| 15–25 dBZ | +3.98 dB |
| 25–35 dBZ | +2.84 dB |
| 35–45 dBZ | +1.86 dB |
| 45–60 dBZ | −5.91 dB |

OPERA's `DBZH` is a maximum over all elevations of all European radars (the CIRRUS product); WN maxes over the German radars' flat precipitation scan, 0.3°–1.8°. Where the near-ground echo is weak the column maximum picks up much stronger signal aloft; where it is strong the maximum already sits near the ground. That is why one exponent cannot fit.

Two independent routes to the class boundaries — via WN plus the documented DWD relation, and via quantile-matching colour areas in the finished image — agree to **1–2.5 dB** across 0.4 to 10 mm/h. They diverge only at the lowest class, where the physics says 11.3 dBZ but the image only starts colouring at 18.8: the DWD's image product clips the weakest echoes, consistent with the "reduzierte Farbklassenauswahl" its brochure describes.

The resulting threshold table, dBZ, one per DWD class:

```
18.77  21.42  21.42  25.95  30.43  34.62  38.68  42.82
47.81  52.91  52.91  56.69  60.70  62.96  66.14
```

## Consequence for the PR

#2998 is being reworked to use **OPERA `DBZH` as the only radar source**, dropping the DWD WMS route: OPERA covers Europe rather than Germany, needs no account either way, resolves finer (1 km against the 3.28 km of the pc_met European image), and with the table above renders in the colours German pilots are used to.

dspreitz, 2026-08-28:
Side by side, so the calibration above is not just a table of numbers. Both frames are **28 Aug 2026, 22:35 UTC**, rendered into the same map section.

**First image — the OPERA composite, rendered by the method described above.** `DBZH` from the 2 km overview level, reprojected from the OPERA Lambert azimuthal grid into the DWD's polar stereographic section, coloured with the DWD's own 15-class ramp using the empirically calibrated reflectivity thresholds. The magenta line marks the edge of radar coverage; the darker grey outside it is where no radar reaches, the lighter grey inside it is covered but without echo. Nothing else is drawn.

[image]

**Second image — the reference: the DWD's "Precipitation Intensity (EuCom)" from pc_met**, which needs a paid subscription.

[image]

## What the comparison shows

**The weather matches.** The cell west of Ireland, the band running from Denmark north-east over the Baltic, the compact complex over southern Germany and Austria, the cells over northern Italy and Corsica — same position, same shape, same intensity gradients.

**The colours match**, which is the point of the calibration. Class histograms over the whole frame agree to −0.2 % in precipitating area and 3.15 against 3.16 in mean class index. Getting there needed more than a Z-R relation: the offset between OPERA's all-elevation maximum and the near-ground product behind EuCom runs from +8.4 dB at weak echo to +1.9 dB at strong, so the class boundaries were measured rather than computed.

**The coverage boundary matches**, scallop for scallop, including the individual radar circles. That is the strongest evidence that the georeference is right — it was derived by fitting the two coverage masks against each other, and the graticule of the reference image confirms it independently: predicted meridians and parallels land on its dash-dot lines.

## What differs, and why that is fine here

The reference carries a base map, borders, rivers, station identifiers, a legend and lightning strikes. The rendering carries none of that, deliberately: XCSoar draws terrain, airspace and the task itself, so an overlay with its own coastlines baked in would sit on top of the map twice. Lightning has no equivalent in OPERA at all; if it is wanted, MTG's `mtg_fd:li_afa` runs at five minutes and would be a separate layer.

The rendering is also the finer of the two. This one is from the 2 km level; the reference image's own raster is 3.28 km/px, and OPERA's full resolution is 1 km.

dspreitz, 2026-10-06:
Closing this, as addressed in #3000

closed 2026-10-06 by dspreitz