XCSoar/XCSoar #175: Flight download doesn't work with IMI Erixx logger

kicune, 2019-03-22:
Due to the change in the newer version of IMI Erixx firmware (version 1.2 onwards, authored in 2015) it is not currently possible to download flights from the device (the request to download the list of flights fails).

The problem is caused by the buffer size in COMM_MAX_PAYLOAD_SIZE in the file src/Device/Driver/IMI/Protocol/Types.hpp, which currently has the value of 1024B. 

The new correct value should be 2048.

kicune, 2019-03-26:
For details and discourse of the topic see also in my previous pull request, that I was not able to get merged: https://github.com/XCSoar/XCSoar/pull/61

DanD222, 2019-04-10:
Change committed to v6.8.x branch https://github.com/XCSoar/XCSoar/commit/458ec04ab8a6b2cf1df3fcfed03ce3e960c5fbed

closed 2019-04-30 by MaxKellermann