XCSoar/XCSoar #178: include average climb vario needle for whole thermal

condoratberlin, 2019-04-08:
It would be great if there could be a  third vario needle indicator available for average climb over the whole actual thermal. may be a 90degree right rotatet "T" to indicate the average climb for the whole thermal (additional to the 30second average). This allows you to see extreamly fast if the 30s average is getting below the overall climb rate.
Additionally the color can indicate the tendency. Green for increasing overall climb rate and red for decreasing.

condoratberlin, 2020-04-23:
Ok, I took a look at C++ (I'm currently only familiar with python) and found a way to integrate this feature.
In my opinion, the filled triangle at the top of the coloured vario bar as can be seen on the following picture is a duplicated information.
[image]
The colour bar is good enough for the current climb value. This filled triangle can now easily be used for an integrated climb speed for the whole current thermal.
I found all necessary information already calculated and stored.
To calculate the current integrated climb speed I would use:

`integrated_thermal=((Basic().nav_altitude + Basic().energy_height) - Calculated().climb_start_altitude_te) / (Basic().time - Calculated().climb_start_time);`

I implemented the calculation value in the GaugeVario.cpp as can be seen on the the attached patch file
[new_vario.zip](https://github.com/XCSoar/XCSoar/files/4524784/new_vario.zip).

The result can be seen here:
[image]
Now you can clearly see, that you are the thermal is getting better and you get the integrated climb of the current thermal.

Now there are a couple of open issues.

1. The calculation should not be performed in GaugeVario.cpp. But there is no obvious point for me where to implement.
2. Further the glide phase is not considered. Here the last integrated climb should be shown (In my opinion)

For these two points I need some help for the right place to implement.
Further this should be only an option, because may be not everybody want to have this vario gauge style.

fb, 2020-04-25:
>  1. The calculation should not be performed in GaugeVario.cpp. But there is no obvious point for me where to implement.

I think you could use `CommonInterface::Calculated().current_thermal`. This is what the *thermal average* infobox uses.

condoratberlin, 2020-04-26:
Oh, great. Thank you for this advise. I didn't found that before. It is exactly what i was looking for. Now only the optional setting is missing.

PR #396 (2020-05-14): fixes #178 added an optional vario needle mode
fixes #178 by showing the average thermal climb ratio instead of current climb ratio. The color bar shows unchanged the current thermal. While not climbing the last average climb ratio is shown. This view allows a fast assessment of the current thermal by comparing the 30s average and the average over thermal.
commit ca041f67 (2020-05-14): fixes #178 added another optional vario needle mode showing the climb rate averaged over the current thermal instead the current climb rate. This option is disabled by default, which ensures an used behaviour.
closed 2020-05-14 by MaxKellermann