XCSoar/XCSoar #180: Why was ios dropped as a supported target

estk, 2019-04-08:
Just wondering why ios was dropped as a supported target, it seems lk8000 has added a build to testflight and it appears to work well. Just wondering why XCSoar hasn't done something similar.

MaxKellermann, 2019-04-08:
Because nobody cares. Meaning: nobody volunteers to maintain this. If you do, join the developer team.

closed 2019-04-08 by MaxKellermann

lordfolken, 2019-04-08:
Nobody maintained it, there was little interest. And somebody would have had to finance the $1000 Apple Developer license just to be able to submit to the the app store. Without that you can only run it on specially enabled  developer phones. Further squashing the user base.

In addition Apple really hates GPL licenced software and has effectively banned it from their store. 

http://meta.ath0.com/2012/02/05/apples-great-gpl-purge/
https://www.engadget.com/2011/01/09/the-gpl-the-app-store-and-you/
https://www.zdnet.com/article/apples-base-strategy-deployed-against-fsf/

Another reason is that you need apple hardware to develop for apple. I'm not willing to furthen a walled garden with my time.

So for all those reasons: get an [openvario](http://openvario.org/)