XCSoar/XCSoar #181: Replace GUI Toolkit with QT

lordfolken, 2019-04-08:
I know this is a long shot, but: 

QT is supported on all the relevant platforms:
* https://doc.qt.io/qt-5/android.html
* https://doc.qt.io/qt-5/linux.html
* https://doc.qt.io/qt-5/windows.html

One of the major complexities is the GUI Toolkit that we currently use. (IMO). Moving to a well known[tm] system would lessen this. 

QT would be attractive to developers and is well understood at this point.

kflog also seems to use qt: http://www.kflog.org/2016/07/cumulus-5-29-1/

MaxKellermann, 2019-04-08:
I agree that it would be advantageous to migrate to a standard GUI toolkit, but I fear that Qt is not a good candidate.
- it requires an obscure preprocessor (moc) which means you're writing moc and not C++
- Qt is a bad citizen in the C++ world, e.g. declaring redundant classes such as `QString` instead of `std::string`, which adds overhead for each roundtrip from C++ and Qt; it has its own idea of multi-threading, instead of using what the C++ standard provides; and much more
- Qt works badly on the Kobo (just compare the performance of the original Kobo Qt-based GUI with XCSoar's software renderer)
- Qt has QML which sounds like a good idea, but it has yet to be demonstrated that it is as powerful as XCSoar's dynamic modal dialog code, which easily adapts to all screen formats and screen densities
- Qt adds a lot of bloat and runtime overhead; XCSoar's static Kobo binary weighs 5 MB (which already includes the whole GUI toolkit, the C++ standard library and the C standard library!), and Qt adds a whopping 25 MB to that, WTF!

Of course, XCSoar's GUI toolkit has large weaknesses (next to its great strengths). And Qt has advantages. Unfortunately, Qt is pretty much the only choice for a cross-platform toolkit. I wish we had better choices.

hor63, 2019-04-08:
Question is: What do we gain using QT, or what are we lacking with what we have.
Is the Staus quo good enough compared the drain on already unsufficient resources to achieve the same with a different toolkit and maybe more contemporary look?
I‘d say: Let‘s choose our battles wisely, and stick in this area with what we have.
Unless someone dedicates him/herself full-time on the topic.

Cheers,
Kai

MaxKellermann, 2019-05-11:
A nice overview: https://philippegroarke.com/posts/2018/c++_ui_solutions/

markus2marks, 2019-10-16:
Is it possible that we use the wxwidget library. Do you have think about it?

MaxKellermann, 2019-10-16:
I always thought of wxWidgets that it combines the worst of all, being just a wrapper for some other native toolkit. On Linux, wxWidgets adds 20 MB, plus a dependency on GTK3 (another 10 MB).
I don't know how good wxWidgets's Android support is, and even less if it works on the Kobo at all. Kobo support is probably the most difficult part; I think no toolkit beats XCSoar's highly optimized code here.

markus2marks, 2019-10-16:
Ok, thank you for your opinion. But the design of the buttons and so on are very bad. With a library like wxwidget its so easy to make it better.

MaxKellermann, 2019-10-16:
> But the design of the buttons and so on are very bad. With a library like wxwidget its so easy to make it better.

Here's the code for rendering a button:
https://github.com/XCSoar/XCSoar/blob/acaa5720f3a25fad2affdf3f06f737a1660c6961/src/Renderer/ButtonRenderer.cpp#L36-L92
Yes, the button looks ugly and easy to make better. But you don't need wxWidgets for this. Just write some code, and submit it as a pull request.

wxWidgets is incapable of rendering to the frame buffer on Linux; it requires X11 or GTK. I don't think it's a good candidate for XCSoar. It would add shitloads of unnecessary bloat. If anybody disagrees - show me some code and demonstrate that it runs well on all devices that Linux supports (Linux, Windows, Android, Kobo, Raspberry Pi with KMS or Wayland, ...). That's a very tough benchmark, but XCSoar's custom toolkit can do that extremely well (but is also extremely ugly).

kubark42, 2020-04-10:
I realize this is an older topic, but I thought it would be worth chiming in. My background is having used Qt extensively, including for a drone Ground Control Station and a sailboat chart application.

> I agree that it would be advantageous to migrate to a standard GUI toolkit, but I fear that Qt is not a good candidate. It requires an obscure preprocessor (moc) which means you're writing moc and not C++.

I don't understand this point. Everything in our GCS was written directly in C++ and QML, we never had a reason to work with the `moc` files. These were automagically generated in the background, much like using CMake or any other build system.

Overall, 

> Qt is a bad citizen in the C++ world, e.g. declaring redundant classes such as QString instead of std::string, which adds overhead for each roundtrip from C++ and Qt; it has its own idea of multi-threading, instead of using what the C++ standard provides; and much more

I would argue that this is a feature, as C++ is not intended to be a multi-platform toolkit. There is a sublime beauty to being able to test your app on your desktop and know that it will look identical on every other platform, including iOS and embedded.

There are of course compromises, but in my experience the time and effort we saved was the so substantial that I don't see how we could have succeeded otherwise. 

> Qt works badly on the Kobo (just compare the performance of the original Kobo Qt-based GUI with XCSoar's software renderer)

That sounds like a blocker. Although Qt works on iOS and embedded, so you'd have to weigh it were worth it to lose a platform in order to gain another.

> Qt has QML which sounds like a good idea, but it has yet to be demonstrated that it is as powerful as XCSoar's dynamic modal dialog code, which easily adapts to all screen formats and screen densities

In my experience, Qt excels at adapting to different screen sizes and densities. That's not to say it's as good as what you already have.

> Qt adds a lot of bloat and runtime overhead; XCSoar's static Kobo binary weighs 5 MB (which already includes the whole GUI toolkit, the C++ standard library and the C standard library!), and Qt adds a whopping 25 MB to that, WTF!

Speaking from an embedded C background, where debates are about saving 4 bytes by using a `float` vs `double`, I never noticed the extra size in the UI. HD space is so much cheaper than developer time, and when 5 pictures on my phone take up 25MB and I get 50MB attachments via email it's hard to make a strong case that 5MB or 25MB is blocking.

FWIW, the binary might grow by even more when using QML, but I have never in my life seen anything as easy to develop as QML. It's almost magical the way the front-end javascript interacts with the backend C++, and it's impossible to overstate the advantage of decoupling front end people-- often time artists and others with limited computer programming background-- from the icky backend.

> Of course, XCSoar's GUI toolkit has large weaknesses (next to its great strengths). And Qt has advantages. Unfortunately, Qt is pretty much the only choice for a cross-platform toolkit. **I wish we had better choices.**

Sigh, me, too. IMHO, the biggest reason not to turn to Qt is the result of a company decision in January to make it much harder to use Qt for dev: https://www.qt.io/blog/qt-offering-changes-2020. The requirements to have an account mean that automated dev installs will die, with the upshot that it's very hard as project maintainers to make it so that everyone is contributing from the same versions. The loss of LTS is heartrending, as that means for safety critical tasks (which I think XCSoar is) that there is a loss of significant control over QC. That's a huge no-no.

Of course, Qt can change course and undo these changes, but the fact that it has tried this is enough to make me think hard about using Qt any further. I know they have bills to pay, but I just don't buy their explanation for how forcing us to have an account in order to download the open-source toolkit is helpful to the community.

There's also [Flutter](https://flutter.dev/), but I don't think you'll get anywhere near the battery performance you get with Qt.

kubark42, 2020-04-10:
One more thing to chime in on (as a separate post because it's a separate notion from the above)

> Question is: What do we gain using QT, or what are we lacking with what we have.

Quite possibly nothing, but it's important not to think of Qt as a GUI framework. It's a cross-platform interface framework, which interfaces as easily with people as with other machines. It has native support for BLE, serial ports, files, networking, maps of all kinds, javascript libs, online stores, user registration, etc... It's a swiss-army knife platform, and it works exceptionally well.

So if your feature roadmap has you moving in lots of directions, all of which are orthogonal to each other, then Qt is immensely valuable. For instance, if you wanted to run a BLE interface with new-fangled wireless sensors, while providing an app which worked on iOS/Android and macOS/Win/Linux/embedded, and not blowing the battery budget, and supporting a variety of map sources, all while uploading user data (e.g. crowdsourced thermal location) in real-time, Qt would be the only way forward (short of having a hugely talented and diverse staff).

If you're focusing on doing one thing and doing it extremely well, then stick with what you've got.

> Is the Staus quo good enough compared the drain on already insufficient resources to achieve the same with a different toolkit and maybe more contemporary look?

Support burden decreases significantly with Qt, both for doing development and bring onboard new devs. We gained significant momentum by lowering the barrier to contributions. People who had the UI/UX skills we needed were able to come up with meaningful code, instead of being stuck at only making feature requests in forums.

> I‘d say: Let‘s choose our battles wisely, and stick in this area with what we have. Unless someone dedicates him/herself full-time on the topic.

Amen.

MaxKellermann, 2020-04-10:
Current Qt company politics means Qt is unacceptable to switch to. Maybe the KDE project decides to do a full-blown fork, that would be perfect. But Qt has lost a huge amount of my trust this week. It is hard to estimate the permanent damage that the Qt company has done to its own product with this move.

lordfolken, 2021-03-27:
So there would be (flutter)[https://flutter.dev]
 - Native binaries (linux,windows,osx,ios,android)
 - C++ bindings (https://flutter.dev/docs/development/platform-integration/c-interop)
 - (Desktop support )[https://flutter.dev/desktop]

scls19fr, 2021-10-03:
Flutter have interessing map widget https://github.com/fleaflet/flutter_map and https://www.syncfusion.com/flutter-widgets/flutter-maps which could be used to draw WMS / WMTS maps as exposed in https://github.com/XCSoar/XCSoar/issues/521