XCSoar/XCSoar #184: pc_met in 7.0_preview13-Android not working

Developer66, 2019-04-12:
In version 7.0_preview13-Android the pc_met access is not working any more. 

- Log give no important informations. 
- The credentials are all correct. 

Seems dwd changed their web structure. In [Image.cpp](https://github.com/XCSoar/XCSoar/blob/master/src/Weather/PCMet/Images.cpp) in src/Weather/PCMet the urls are not working by copying them manually in the browser.

- [xcsoar.log](https://github.com/XCSoar/XCSoar/files/3073531/xcsoar.log)
- [Screenshot_XCSoar](https://user-images.githubusercontent.com/14195975/56042070-7ab9e400-5d3a-11e9-88c9-30e7835f1158.jpg)

Any help would be greatly appreciated! Cheers :)

dspreitz, 2019-04-12:
I believe DWD changed the way they distribute data due to the new DWD law passed in Germany.
See #174 dor more details.
On the upsite, DWD is now offering geo referenced images, which in the past they didn't do, AFAIK.

Developer66, 2019-04-12:
I am looking forward to the future :)

kerel-fs, 2019-09-18:
The flugwetter http web server returns HTTP 302 Found / URL redirection to the TLS-encrypted https://flugwetter.de when accessing one of the resources used by XCSoar, e.g.
<http://flugwetter.de/fw/bilder/rad_lokal/einzelstandorte.htm?type=pro> .

The path for local radar images the paths didn't change (I didn't check the other resources).

So I suspect that either XCSoar doesn't follow the redirect or can not access TLS-encrypted resources at all. Accessing the https server directly by changing the `PCMET_URI` would probably fix this issue.

Paraendurist, 2021-02-21:
Unfortunately this Problem persits in v15. Did anybody find out the reason for the Bug by the time.

Paraendurist, 2021-02-27:
I accsessed the dwd Support and easylygot a test Account. Unfortunately xcsoar 7.15 did not work. I am sorry i cant help in programming but if someone is able to Do that dwd is open to Support US.

dpluigi, 2021-03-15:
Hi I am just starting to look at the overlay feature in XCSoar 7 (preview) and I have a DWD free account is it enough to provide basic 3 day forcast and overlays?  What is the FTP PC-Met account in the XCSoar Weather setting?  

On a separate note I was able to overlay Skysight thanks to Caz Yokoyama:   https://www.youtube.com/watch?v=I52nFW3i4m0&lc=UgyjhXhLVBQj-4lXR214AaABAg.9JJkPC60EJC9Ksm1d9I5m1

MaxKellermann, 2021-03-15:
@dpluigi this is about a different feature, not the pc_met overlays. And this is certainly not about Skysight.

MaxKellermann, 2021-03-15:
Fixed by ff5e107f5db42bfd0bc8acb397b4f296b9ec844c

closed 2021-03-15 by MaxKellermann

dpluigi, 2021-03-15:
> @dpluigi this is about a different feature, not the pc_met overlays. And this is certainly not about Skysight.
@MaxKellermann 
My bad, I thought it was about PC-Met (from a keyword search).  I commented on Skysight by passing and on a realted interest.  I  am now starting to scratch the many features of XCSoar as I am starting to spread my wings from local flights since I solo-ed to Campagne training this coming season.
Thank you for all the good work.

MaxKellermann, 2021-03-15:
>  I commented on Skysight by passing and on a realted interest.

Note that @CazYokoyama has abandoned the SkySight code. He submitted his code in 2019, but was not willing to finish the work to meet XCSoar's quality requirements. See https://github.com/XCSoar/XCSoar/pull/298#issuecomment-599304860
That is very unfortunate.

Paraendurist, 2021-03-15:
Hello we are meanwhile on preview 15. So I think there is nö use implementing in 13. But I realy appreciate your work. The dwd guys told ne there is nö cintract anymore between xcsoar and dwd. But I dont know weather there has been any

⁣Mit freundlichen Grüßen 

Peter M. Groß

Dipl. Oec. Mediator Cert. 
Manager of Company Networks
Tel: +49 (07622) 6726311
Tel: +49 (0152) 29304000
Drosselweg 5
D-79650 Schopfheim

Gesendet mit TypeApp ​

https://github.com/XCSoar/XCSoar/issues/184#issuecomment-799517245

dpluigi, 2021-03-15:
> > I commented on Skysight by passing and on a related interest.
> 
> Note that @CazYokoyama has abandoned the SkySight code. He submitted his code in 2019, but was not willing to finish the work to meet XCSoar's quality requirements. See [#298 (comment)](https://github.com/XCSoar/XCSoar/pull/298#issuecomment-599304860)
> That is very unfortunate.

This is unfortunate indeed, hopefully that there will be someone who can finish and clean the code started by Caz.  Thank you for the clarification and the link to thread #298, I am not a programmer, I just tinkered as a grad student some quick and dirty C and C++ coding many moons ago.  I will ask around the gliding community in France if anyone is willing, there is a big demand.  

Thank you for the quality of the XCSoar implementations.