XCSoar/XCSoar #192: Builds are broken

Turbo87, 2019-05-06:
It seems that between 6412cdf373f59f627f78fd8751c098ae7d45ea91 and 439cd89c103b325f9f88338641c677ae1cd475cd the CI builds started failing.

From the build log:

```
output/UNIX/dbg/driver.a(KRT2.o): In function `KRT2Device::Send(unsigned char const*, unsigned int, OperationEnvironment&)':
/root/project/src/Device/Driver/KRT2.cpp:181: undefined reference to `KRT2Device::CMD_TIMEOUT'
```

/cc @MaxKellermann

kerel-fs, 2019-05-06:
Commit 439cd89c103b325f9f88338641c677ae1cd475cd changed the variable in question.

MaxKellermann, 2019-05-06:
This was already bugged, but the bug now materializes because the compiler chose to refer to the variable by reference, not by value after that commit.

MaxKellermann, 2019-05-06:
Actually, I just found out that this is a compiler bug, see https://en.cppreference.com/w/cpp/language/static
> If a static data member is declared constexpr, it is implicitly inline and does not need to be redeclared at namespace scope. This redeclaration without an initializer (formerly required as shown above) is still permitted, but is deprecated.

Turbo87, 2019-05-06:
@MaxKellermann so CI won't be fixed because it's a compiler bug?

MaxKellermann, 2019-05-06:
I don't understand the question. Maybe that answers your question: XCSoar doesn't need to be fixed because it isn't broken.

Turbo87, 2019-05-06:
well, XCSoar's CI is broken, and IMHO that needs fixing. I think https://github.com/XCSoar/XCSoar/commit/439cd89c103b325f9f88338641c677ae1cd475cd shouldn't have been merged yet until we could update CI to use a compiler without that bug.

MaxKellermann, 2019-05-06:
CI is a useful tool to measure correctness and detect errors, but if the tool turns out to be flawed, we don't need to listen to it. Therefore I disagree that it shouldn't have been merged. A broken tool shouldn't be able to delay development.

kerel-fs, 2019-05-07:
Fixed in master.

closed 2019-05-07 by kerel-fs