XCSoar/XCSoar #197: Compilation issues on v6.8.x branch (target Kobo)

nico-krais, 2019-05-13:
There are some issues when trying to compile the v6.8.x branch for the Kobo. This is on an Ubuntu 18.04 system, using the arm-linux-gnueabihf-gcc-5 in version 5.5.0, with the debug options active.

* The glibc 2.23 that is used seems to be bugged, see https://sourceware.org/git/gitweb.cgi?p=glibc.git;a=commitdiff;h=388b4f1a02f3a801965028bbfcd48d905638b797 for details. A solution is to patch the glibc before compiling in the way mentioned there. I patched a version and uploaded it to github, see https://github.com/nico-krais/XCSoar/commit/3bb96fff35b504e8b396a391db7bc13badbd2ab6

Error message example:

```
/tmp/ccWYEnuM.s: Assembler messages:
/tmp/ccWYEnuM.s: Error: `loc1@GLIBC_2.4' can't be versioned to common symbol 'loc1'
/tmp/ccWYEnuM.s: Error: `loc2@GLIBC_2.4' can't be versioned to common symbol 'loc2'
/tmp/ccWYEnuM.s: Error: `locs@GLIBC_2.4' can't be versioned to common symbol 'locs'
../o-iterator.mk:9: recipe for target '/home/xxx/codes/XCSoar/output/KOBO/lib/arm-linux-gnueabihf/build/glibc-2.23/misc/regexp.os' failed
```

* An include is missing in `src/Event/Poll/Linux/Input.cpp`, meaning compilation will abort. Fix see https://github.com/nico-krais/XCSoar/commit/d1789700cc5806d66b38ece6c67de171fd202326.

Error message example:

```
src/Event/Poll/Linux/Input.cpp: In member function ‘void LinuxInputDevice::Read()’:
src/Event/Poll/Linux/Input.cpp:133:9: error: ‘errno’ was not declared in this scope
     if (errno != EAGAIN && errno != EINTR)
         ^
src/Event/Poll/Linux/Input.cpp:133:18: error: ‘EAGAIN’ was not declared in this scope
     if (errno != EAGAIN && errno != EINTR)
                  ^
src/Event/Poll/Linux/Input.cpp:133:37: error: ‘EINTR’ was not declared in this scope
     if (errno != EAGAIN && errno != EINTR)
                                     ^
build/compile.mk:114: recipe for target 'output/KOBO/dbg/src/Event/Poll/Linux/Input.o' failed
```

kerel-fs, 2019-09-18:
>    The glibc 2.23 that is used seems to be bugged, see https://sourceware.org/git/gitweb.cgi?p=glibc.git;a=commitdiff;h=388b4f1a02f3a801965028bbfcd48d905638b797 for details. A solution is to patch the glibc before compiling in the way mentioned there. I patched a version and uploaded it to github, see nico-krais@3bb96ff

A patch solving this issue was added to xcsoar in 3ea0e52 but never backported to the v6.x branch.
Would you like to backport it? I'm not able to build v6.x yet.

Regarding your second issue: It looks good to me, but I'm wondering why this didn't break the build previously / for other targets. If you still think this is a valid patch it would be great if you could rebase it on v6.x and submit a Pull Request!

Sincerely,
kerel

MaxKellermann, 2020-06-13:
I have backported the Musl patches to v6.8.x in 6.8.13, so this issue should be gone.

closed 2020-06-13 by MaxKellermann