XCSoar/XCSoar #200: Airspace warning shouldn't replace bottom area

zinuzoid, 2019-05-19:
Currently, Airspace warning dialog replace the bottom area (in this case Cross section). Ideally this shouldn't replace but show it on above or on top of the bottom area.

Before Airspace warning:
[image]

Airspace warning appear and Cross section disappear:
[image]

After acknowledge Airspace warning, Cross section is back:
[image]

Since I'm also a developer I would be happy to get some code pointer so I can help contribute to this.

MaxKellermann, 2019-05-20:
The framework currently allows one "widget" to be placed below the map, which can be the cross section, for example. Airspace warnings use the same API, so they replace the cross section widget. That part needs to be made stackable, probably.

zinuzoid, 2019-05-20:
Hmm, sounds like a big changes. Is there anything I can help contribute then?

lordfolken, 2023-03-22:
Can't we make it a window? And artificially move it to the bottom of the screen? 
This would solve the stacking and also the input/focus problem. 
Because a further problem with this popup is that there is no keyboard focus for it. You can ack it with enter but only via the xci file and the input event subsystem.

closed 2026-10-05 by lordfolken