XCSoar/XCSoar #201: Supporting GDL90

dgofman, 2019-05-20:
Hello guys,
Please, can you support GDL90 interface? We have a problem getting ADS-B traffic data on XCSoar.
Thanks

https://github.com/PepperJo/stratux/issues/4

MaxKellermann, 2019-05-20:
What is GDL90?

dgofman, 2019-05-20:
A library :) https://github.com/etdey/gdl90
As I believe a Garmin interface
https://www.faa.gov/nextgen/programs/adsb/Archival/media/GDL90_Public_ICD_RevA.PDF

MaxKellermann, 2019-05-20:
That's a Python library. XCSoar is C++. Not compatible.

dgofman, 2019-05-20:
You can use C++ code

https://github.com/search?l=C%2B%2B&q=GDL90&type=Code

for example: https://github.com/XenonofArcticus/libGDL90

dgofman, 2019-05-20:
Looks like that library already integrated with XCSoar

https://github.com/lyusupov/SoftRF

https://github.com/lyusupov/SoftRF/raw/master/documents/images/softrf-emulation.jpg

MaxKellermann, 2019-05-20:
> for example: https://github.com/XenonofArcticus/libGDL90

After looking at the code for 1 minute, I found several buffer overflows. Crap code quality.

> Looks like that library already integrated with XCSoar

No.

MaxKellermann, 2019-05-20:
> https://github.com/lyusupov/SoftRF

This repository contains code to generate GDL90, not to parse it. And its code quality is ugly as well; it uses global static buffers. Even if it would implement what we need, its code quality is unacceptable for XCSoar.

dgofman, 2019-05-20:
The best implementation I found on Python (clean code)
https://github.com/etdey/gdl90/tree/master/gdl90
Another implementation using  Objective C
https://github.com/Carthage/Carthage/issues/708   however the source code
it's not available anymore.

dgofman, 2019-05-20:
Please review this post as well.
https://github.com/lyusupov/SoftRF/issues/31

dgofman, 2019-05-20:
I believe you need only one class for review
https://github.com/lyusupov/SoftRF/blob/master/software/firmware/source/SoftRF/GDL90Helper.cpp

hor63, 2019-05-24:
Why Reviewing? You must implement a driver for XCSoar. This class is for SoftRF.
BTW: Copy and Paste ist not permitted. The file is under GPL V3. XCSoar is GPL V2. You can incorporate GPL V2 code into a GPL V3 project, but not vice versa. See [GPL-Compatible Free Software Licenses](https://www.gnu.org/licenses/license-list.en.html#GPLCompatibleLicenses).
You may let yourself insipre by the code but copying it verbatim is a nono. You must re-write it from scratch.

MaxKellermann, 2019-05-24:
XCSoar is GPLv2 or later, so using GPLv3 code would be legal, but it would force XCSoar to switch to GPLv3 or later. That would be fine for me.

hor63, 2019-05-24:
> A library :) https://github.com/etdey/gdl90
> As I believe a Garmin interface https://www.faa.gov/nextgen/programs/adsb/Archival/media/GDL90_Public_ICD_RevA.PDF

Is there a concrete device or project which you want to connect to? 
According to Garmin's web site the GDL 90 device is out of production already.
Are there any products which use this protocol? If so most of the garmin stuff is for GA but not for glider operations.

Plantain, 2019-05-24:
He mentioned his intentions in the original post - https://github.com/PepperJo/stratux/issues/4

hor63, 2019-05-24:
Thanks. That make a lot more sense now.

dgofman, 2019-05-25:
I found two GIT repositories one of them for decoding FLARM  https://github.com/creaktive/flare
another is implementation on the statux device. https://github.com/0x74-0x62/stratux/tree/devel/flarm_receiver
Now using the same code you can implement startux2FLARM Receiving both ADS-B on 1090 MHz and FLARM on 868 MHz.

kerel-fs, 2019-09-18:
As already described in https://github.com/XCSoar/XCSoar/issues/201#issuecomment-495549669 it would make much more sense to implement an NMEA output in stratux to achieve the desired interoperability, as already proposed in https://github.com/cyoung/stratux/issues/735 and https://github.com/cyoung/stratux/issues/741.

closed 2019-09-18 by kerel-fs

scls19fr, 2021-10-03:
""SafeSky"" https://www.safesky.app/ seems to also support GDL90 in their "early birds" subscription see forum discussion https://forum.safesky.app/t/integration-safesky-sdvfr/275 and blog post https://www.safesky.app/fr/news/interconnectivite-et-protocole-gdl90-id42
It could help to provide in XCSoar several aircrafts trafic (SafeSky, ADS-B, FLARM and OGN) and so possibly improve a bit safety.

lordfolken, 2021-10-05:
> SafeSky https://www.safesky.app/ seems to also support GDL90 in their "early birds" subscription See forum discussion https://forum.safesky.app/t/integration-safesky-sdvfr/275 and blog post https://www.safesky.app/fr/news/interconnectivite-et-protocole-gdl90-id42 It could help to provide in XCSoar several aircrafts traffic (SafeSky, ADS-B, FLARM and OGN) and so possibly improve a bit safety.
Powerflarm does already do sensor fusion for ads-b, transponder mode-c and flarm. XCSoar support has support for skylines's friends protocol if you want to do "over the horizon" tracking. 
Nevertheless i found out today, that PowerFLARM Fusion also supports GDL-90 as an output. Traffic warnings are less sophisticated on GDL-90 then on flarm (only one alert stage). The main advantage of the Protocol is that it will relay things like weather radar maps and far away traffic via TIS-B. This is however sadly only supported in the US.

dgofman, 2021-10-05:
@lordfolken I believe it's working only in Europe