XCSoar/XCSoar #203: Website is down?

Symbian9, 2019-05-20:
https://www.xcsoar.org/ returns "Error 443"

> ### Forbidden
> 
> You don't have permission to access / on this server.
> 
> ***
> 
> Apache/2.4.25 (Debian) Server at www.xcsoar.org Port 443

mongrelion, 2019-05-21:
It's not down. It loads over here.

Symbian9, 2019-05-21:
> It's not down. It loads over here.

For me it still unavailable and returns "Error 403"

[image]

Also just tested http://www.xcsoar.org/ using "Site24x7" and result is next:  ***"This url is forbidden."***

- http://www.site24x7.com/public/t/results-1558444840827.html

hor63, 2019-05-24:
Could it be that your provider/company/router... blocks the access to the IP address?
First week of May the site was indeed down due to a provider change on rather short notice.
But it is up since.

hor63, 2019-05-24:
Try:
`ping xcsoar.org`

Output for me is: 

> Pinging xcsoar.org [18.184.238.205] with 32 bytes of data:

If ping or nslookup shows a different address your name server still has an outdated IP address or a man-in-the-middle tries to intercept and manipulate traffic.

Symbian9, 2019-05-24:
> Try:
>
> `ping xcsoar.org`

Here is output:

```
$ ping xcsoar.org
PING xcsoar.org (18.184.238.205) 56(84) bytes of data.
```

hor63, 2019-05-24:
Seems to be specific for you. The site is up running and reachable.

Symbian9, 2019-05-24:
BTW, could you create mirror as:

- https://xcsoar.github.io

By creating repo

- https://github.com/xcsoar/xcsoar.github.io

Also, please, upload binary as assets to releases

- https://github.com/xcsoar/xcsoar/releases

Plantain, 2019-05-24:
Clearly this is not some ISP blocking as his screenshot shows a successful
SSL connection and a 403 served from XCsoar.org. I suspect the new website
suffers from the same overzealous banning problems as the forum.

lordfolken, 2019-05-27:
I have now corrected the webserver config. Only forum.xcsoar.org is now being filtered. @Symbian9 could you validate and close this bug report if you are able to access the site?

lordfolken, 2019-05-27:
I have created a new issues for the releases. This is planned.

Symbian9, 2019-05-27:
> could you validate and close this bug report if you are able to access the site?

Yes, it is working now!

closed 2019-05-27 by Symbian9