XCSoar/XCSoar #204: Request for semicircle turnpoint sector (or generic sector) of arbitrary width.

hess8, 2019-05-20:
In our group we're doing a new kind of task that requires passing  "goal line" before the end of the flight.  For example see "One-line OLC+ Flight" on www.teamxc.us/how-we-fly (2nd graphic).  In XCSoar we could easily use the Racing task type if a "Semicircle" option were added.   It would be oriented perpendicular to the line leading to it.

Alternatively, a "Generic" turnpoint sector would work great:  arbitrary angle width of sector, arbitrary orientation to line leading to the sector, and arbitrary width.  

Thanks, this would add great flexibility to create new kinds of tasks and competitions.

Exadios, 2019-05-21:
There is currently a turnpoint that is very close to what you want. It is called the "Variable Sector". The only problem is that it generates a "nipple" of at least 100 meters at the center of the line. This is because, currently, it is not possible to enter zero meters for the "Inner Radius".

This turn point is not currently available in the standard XCSoar edition but is available in the edition released by our club - the so called "Narrogin Gliding Club Fork" - on the Android Playstore.

hess8, 2019-05-22:
Thanks.  I guess that's two votes for including this in the main version of
XCSoar.

lordfolken, 2022-05-20:
Can you please provide a graphic or similar?

hess8, 2022-05-21:
Thanks for your work on XCSoar!  You can close this.   200 nmi is wide
enough.

lordfolken, 2022-05-21:
Allright.

closed 2022-05-21 by lordfolken