XCSoar/XCSoar #206: detail files not showing

gvuissoz, 2019-05-24:
Hello, since last update I have an issue with the files attached to landing places, they don't open anymore when I go under details. I went back to 6.8.11 and there is no issue. It seems to be related to the last version.

hor63, 2019-05-24:
Can you elaborate? Which file exactly do you mean?
Do other files load like maps, airspace, topography?
Which OS are you using?

gvuissoz, 2019-05-24:
i mean custom pdf files (approach maps, emergency fields) they are listed in landing.txt and when I want I can go under details and they open in a pdf viewer
Android 8.0.0
xcsoar 6.8.12

kerel-fs, 2019-09-18:
@gvuissoz In which version did you see such a feature (load custom pdf files for waypoints)?
I can't find hints about such a feature in neither the old version v6.8.12 nor the upcoming v7.x.

gvuissoz, 2019-09-20:
Hello,

i used the information I found here:
http://www.volavoile.net/index.php?showtopic=11715
applicable to attach a pdf or other file to the details page of a landing
area.

Best regards

kerel-fs, 2019-09-20:
Hi,
thanks for the link, I didn't know this feature until now (meanwhile I've also read it's description in section 14.5 "Aircraft details" of the manual).

On my Android device (Android 8.1.0) the feature is working without any problems using XCSoar version 6.8.12 as well as a recent build from master ([v7.0_preview14-Android~git#74fcee3](https://gitlab.com/kerel-fs/xcsoar/commit/74fcee351244611b2498316c488710228ad1c127)).

In order to test the feature, I used the [Germany.cup](http://download.xcsoar.org/waypoints/Germany.cup) waypoint files, and the following waypoint details file (`waypoint_details.txt`):
```
[Koln-Bonn]
Lorem ipsum dolor sit amet, consectetur
adipiscing elit, sed do eiusmod tempor
incididunt ut labore et dolore magna
aliqua. Ut enim ad minim veniam, quis
nostrud exercitation ullamco laboris nisi
ut aliquip ex ea commodo consequat.
Duis aute irure dolor in reprehenderit
in voluptate velit esse cillum dolore eu
fugiat nulla pariatur.

Excepteur sint occaecat cupidatat non
proident, sunt in culpa qui officia
deserunt mollit anim id est laborum.
image=eddktest.png
```
as well as a small arbitrary png image `eddktest.png`.

The result looks like the following:
[image]

@gvuissoz Can you provide the files (waypoint & waypoint details) which trigger this issue for you?

gvuissoz, 2019-09-21:
thank you for answer, so far I understand my issue may be related to the
xcsoar files being used from the sd card. Next winter I will install the
6.8.12 and try with different folder location.

kerel-fs, 2019-09-21:
Ok, then I'll close the issue for now.
Feel free to re-open it when you were able to reproduce this issue!

closed 2019-09-21 by kerel-fs