XCSoar/XCSoar #208: no WIFI on kobo aura edition 2

Jeripimo, 2019-05-27:
Hi,

I bought a kobo aura edition 2 last week.
Installed xcsoar 6.8.11

Tapping 'Network" 'button' (before starting xcsoar on main screen) display shows: 'WIFI ON' but in fact the wifi is not working. When pressed whole line turns blacks for a couple of seconds but wifi still not working.
Second line 'WiFi' is greyed out.

White led keeps flashing

Kobo runs the latest firmware.

Regards, Jerome

pyctam, 2019-07-30:
I also have same issue on XCSoar v6.8.11-Kobo. WiFi works with no any problem under the Nickel.

CazYokoyama, 2019-09-08:
I have the same symptom. After I look at source code of xcsoar and do several attempts. I conclude the driver module of wifi chip does not work with the kernel xcsoar provides. As a result, I enables USB-Ethernet device such as Trendnet TU2-ET100 and have happy debugging. caz at caztech dot com

lordfolken, 2019-09-09:
https://github.com/XCSoar/XCSoar/blob/master/kobo/kernel/aura2.config Is the kernel configuration for the KOBO Aura 2. Is the driver missing, or is just not compiled in?

Jeripimo, 2019-09-09:
Hi Phillip,

To be honest I really don't know.

How can I compile it?

Is there a tutorial for how to compile?
I'd like to contribute but don't know how.

Regards,

Jerome

CazYokoyama, 2019-09-09:
The driver is not compiled when kernel is compiled. XCSoar uses the driver which comes with Kobo.

Jeripimo, 2019-09-09:
So is it possible to add another driver for xcsoar to use?

Fuchur777, 2020-03-24:
Just installed the latest version and have the same bug.

dr-raylewis, 2020-05-15:
I have a Kobo Touch 2 working with XCSoar 6.8.13 - there is a fatal bug in this compilation for Kobo that prevents the use of the Kobo WiFi for transfer of data to XCSoar over WiFi from Condor or other source of NMEA data.

As previously reported, looking in network shows wifi to be 'on' but the second button is greyed out. Switch WiFi off and then use the second button to select the correct WiFi network, and you will find 'WiFi on' and the second button still active. This is unexpected behaviour.

The port properly assigned shows 'Not Connected' and will not connect when the 'Reconnect' button is pressed. Go to monitor and there is no stream. From Monitor, try 'Reconnect' and it displays the significant error message 'Device is Occupied' which of course is fatal.

When I quit XCSoar and go to Nickel then I find the WiFi is not automatically connected. I have to manually select the network I want and re-supply the password (which I have previously saved) to make the connection. Sometimes I can reconnect, but most times I have to turn off kobo and switch it on again, going straight to Nickel without running XCSoar. WiFi is disconnected, but will reconnect (without supplying the password) automatically to the last used network if I press the WiFi icon at the top. This would normally happen automatically when switching on, so it seems the XCSoar installation is interfering with the WiFi without switching it on properly.

There is no way (I have found) to go back from Nickel to XCSoar with the WiFi properly connected, nor any way I have found in XCSoar to make the WiFi connection.

I am convinced that this is an issue with XCSoar software and nothing directly to do with Kobo; as an experiment I loaded the Kobo version of LK8000: this allows me to configure and use (n LK8000) the wireless connection for taking NMEA data from Condor!

Jeripimo, 2022-07-06:
Hi,

I found this. Maybe that's the solution for the no wifi issue?

https://github.com/kobolabs/Kobo-Reader/blob/9e4eb089e4d31855a64ff022117b965562eb7ce9/hw/imx6sll-aurah2o2-aura/wifi.tgz

Jeripimo, 2022-07-27:
I have additional info.

Opened kobo and found this wifi driver chip:  Realtek 8189FTV

Maybe someone can try to implement this in the kobo build.

Thanks

lordfolken, 2023-02-26:
@XCSoar/kobo-team anyone have such a device, so maybe its possible to fix the issue?

closed 2025-02-10 by lordfolken