XCSoar/XCSoar #215: Sound for KOBO reader

Jeripimo, 2019-06-09:
Add sound out via USB so a USB audio dongle can be used

lordfolken, 2019-06-20:
Please modify the kernel configuration files for your platform:
https://github.com/XCSoar/XCSoar/tree/master/kobo/kernel

Test and send a pull request to the project.

lordfolken, 2019-06-20:
Are you using the OTG Kernel variant? Because in principal its enabled:

https://github.com/XCSoar/XCSoar/blob/master/kobo/kernel/otg.config#L1156

of course you need also the relevant driver for your specific device.

GeekGoneOld, 2022-02-08:
I can confirm that sound works perfectly on my Kobo Clara HD.  As @lordfolken said, the problem may be specific to his Kobo model or his sound device.  Please indicate these two models.

The "fix" is to ensure that the OTG kernel for your model supports his sound device and if it doesn't, either make a new kernel or make a single module, load it with FTP and make a XCSoarData/kobo/init.sh file to insmod it.

lordfolken, 2026-10-18:
I'm closing this. No answer and not principally broken.

closed 2026-10-18 by lordfolken

GeekGoneOld, 2026-10-19:
Note that sound devices CAN be used through USBOTG for ClaraHD, Libra2 and LibraH2O.  I've tested it on ClaraHD and it worked.  Specific drivers are identified in Kobo/rcS where they are loaded.