XCSoar/XCSoar #216: New vario trail type for black/white screen (kobo)

Jeripimo, 2019-06-09:
Since the kobo doesn't have a color screen it would be nice to have an extra vario trail type.

From small dotted line when sinking to a line that gets wider the more you rise
or
Small open dots when sinking to big filled dot when rising, size of the dots when rising depending on how fast you rise .

So  would be better visible on a black/white screen.

cefn, 2020-07-01:
Hi, @Jerimipo is this already fulfilled by the alternatives to select the following under `Menu => Config => System => Map Display => Elements => Trail Type`

* `Vario #1 (with dots)` which has varying width and dots ?
* `Vario-scaled dots and lines` which has scaled dots?

What would the differences be relative to these pre-existing schemes?

lordfolken, 2021-04-08:
@Jeripimo does @cefn suggestion help you?

cefn, 2021-04-08:
I asked because I was working on an alternative display (which I flew with last season and worked pretty nicely). I was wondering what features others were thinking of with a B+W version... 
[image]

cefn, 2021-04-08:
If I recall it was mostly https://github.com/cefn/XCSoar/blob/74d6a5ae63a89146965f6709a6a2acc26559c71c/src/Look/TrailLook.cpp#L58-L70 combined with some scaling interventions to make the circles bigger according to lift, although I think I may have overdone the scaling a bit https://github.com/cefn/XCSoar/blob/74d6a5ae63a89146965f6709a6a2acc26559c71c/src/Look/TrailLook.cpp#L120

Basically it's the changes in https://github.com/cefn/XCSoar/compare/v6.8.x...74d6a5ae63a89146965f6709a6a2acc26559c71c given I'm Paragliding I also forced the drawing of the next-target line to be off.

lordfolken, 2022-06-16:
@ulipo it looks like a good change. Adding this could work. What we have to work out though is how to make it default when a kobo is detected.

Jeripimo, 2022-07-28:
Looks a good change indeed. Thanks

ulipo, 2022-07-28:
> @ulipo it looks like a good change. Adding this could work. What we have to work out though is how to make it default when a kobo is detected.

I haven't the slightest idea of how to do that, but in the meanwhile that someone more skilled comes up with a solution, we could just add the trail type to be selected by hand...

Let me know if it is ok to make a PR or if it risks to waste precious developpers time.

lordfolken, 2023-02-26:
Why don't you provide a PR. 
A PR should be:
 * atomic 
 * change one thing and one thing only
 * no whitespace or tabs etc. 
 * adhere to code style 
 * commit message in the format: 
    component: text max char 80
    the rest of the text if needed

ulipo, 2023-02-26:
Dunno about the "atomic" part, but I hope I've nailed the rest :-)

Jeripimo, 2023-02-27:
Sorry really don;t know what you mean bij PR

closed 2023-03-01 by MaxKellermann