XCSoar/XCSoar #217: add copilot to logger settings

Luap-Paul, 2019-06-09:
Hello, 
For national records in a double seater its necessary to declerate the logger with the copilot. Because i know pilots, missing this i started to get into the code and try to add it.
I added the copilot to the logger settings and all the related src files, it seems to work. Because i want to declare a nano logger with xcsoar the lx Driver has to be adapted too. At one point the driver already contains a function call to declare "HFCM2CREW2:". My question is, is it possible to adjust the driver for this copilot loggersetting without knowledge (help) of LX. Im new in c# coding and therefore sorry if i got something wrong. If anybody could give me a hint or some help would be great. 
Greetings Paul

lordfolken, 2019-06-20:
Hi!

- Please make a pull request with your code. Add (WIP) so its obvious that it isn't finished. 

Seeing what we are discussing is easier. 

- Modifying the LX Driver
I assume the NANO is behind the LX? If so then the LX will switch to the so called passthrough mode. This talks directly to the nano. The LX is then oblivious to any commands you send to the logger.

Luap-Paul, 2019-06-29:
Thank you for your reply!
The DataPort Specification really helped me.
Today I tested the code with my nano and it worked fine. So as soon i have the time, I will send a pull request, propably next week. 
Greetings Paul

closed 2020-04-15 by Luap-Paul