XCSoar/XCSoar #228: Compilation issues for target Linux

Vitalhb, 2019-07-19:
Linux flavor: Ubuntu 18.04
Issues: Missing libraries on development manual

**1- quilt not found**
~/XCSoar/XCSoar$ make
  GET     output/download/boost_1_70_0.tar.bz2
download https://dl.bintray.com/boostorg/release/1.70.0/source/boost_1_70_0.tar.bz2
  UNTAR   boost_1_70_0.tar.bz2
/bin/sh: 1: quilt: not found
build/libboost.mk:18: recipe for target 'output/src/stamp-boost_1_70_0' failed
make: *** [output/src/stamp-boost_1_70_0] Error 127

Solved by installing quilt: sudo apt-get install quilt

**2- lua5.2 library not found**

Package lua5.2 was not found in the pkg-config search path.
Perhaps you should add the directory containing `lua5.2.pc'
to the PKG_CONFIG_PATH environment variable
No package 'lua5.2' found
build/compile.mk:107: *** library not found: lua5.2.  Stop.

Solved by installing liblua5.2: sudo apt-get install liblua5.2-dev

**3- missing library libtiff-4**

Package libtiff-4 was not found in the pkg-config search path.
Perhaps you should add the directory containing `libtiff-4.pc'
to the PKG_CONFIG_PATH environment variable
No package 'libtiff-4' found
build/compile.mk:107: *** library not found: libtiff-4.  Stop.

Not solved yet.

Please, update the development manual to include all the necessary libraries to build XCSoar for Linux.

lordfolken, 2019-07-22:
Hi!

You gotta be a lot more detailed than that.

First of all Debian is the build target not Ubuntu. 
Logs, packages you installed, and didn't install..  What worked, what didn't. 
There is https://github.com/XCSoar/XCSoar/blob/master/ide/provisioning/install-debian-packages.sh

Which will install the packages needed. 
There is a vagrant box in https://github.com/XCSoar/XCSoar/tree/master/ide/vagrant
And there is my unofficial docker container: 
https://github.com/XCSoar/XCSoar/tree/master/ide/vagrant

If you feel the documentation is inadequate, please update it. The developer is you.

MaxKellermann, 2020-01-15:
Closing due to lack of feedback from submitter.

closed 2020-01-15 by MaxKellermann