XCSoar/XCSoar #230: "Round" scale for maps

scls19fr, 2019-08-03:
Hello,

I'm trying several softwares for gliding and found XCSoar very interesting and quite convivial.
But I noticed something quite odd...

Scale of map is not a round number ie 1.1km, 2.2km, 4.4km, 6,7km, 11km, 17km, 22km, 33km, 44km, 67km, 111km, 222km

why not fixing round scale such as  1.0km, 2.0km, 5.0km, 7.5km, 10km, 20km, 50km, 75km, 100km, 200k...

Your opinion?

Kind regards

scls19fr, 2019-08-10:
An other comment about scale and XCSoar...

[image]

The scale symbol currently represents full screen width.

[image]

It's as if scale on a paper map was given for full map width... you can probably accept that it's not a common practice!

Maybe scale symbol should have a length which is only a percentage of screen width.

Maybe having both scale behavior could be considered (to avoid to disturb users who prefer current scale behavior)

Exadios, 2019-08-11:
So, I've still to see any argument (convincing or otherwise) that supports any change.

scls19fr, 2019-08-11:
## Why round scale?

If you want to quickly measure distance on screen and noticed that distance between 2 points is about a fourth of screen width, it's much quicker and simpler to calculate 200 / 4 than 159 / 4.

## Why scale of a fixed length ?

if scale was of fixed length (and not relative to screen width) and you quickly notice that distance between 2 points is around 4 times length of scale length (with a 20 km scale) you can do 20 *  4 = 80 km

Multiplication is an even easier mental operation than division.

Exadios, 2019-08-15:
So, I don't think I've ever tried to estimate a distance from the screen. How often does this situation arises?

scls19fr, 2019-08-15:
At least 4 or 5 times per flight when considering which turn point I should choose for a free flight (not a declared circuit) given current altitude, position, time and weather condition.

Exadios, 2019-08-16:
Well, it seems to me, that turnspoints only have significance in the context of tasks. Otherwise, a turnpoint is just an arbitrary point which has no more significance than the point directly below you.

I can't remember how XCSoar implements rendering but often times only certain scale factors (e.g. powers of 2) are implemented. If this is the case in XCSoar then the rendering would need to be reworked. Given the number and diversity of XCSoar target hardware this is not a trivial change.

lordfolken, 2022-05-20:
I agree the scale is unclear. 
The scale has to be indicated somehow and understandably. Please discuss or propose a way of implementation.

groundhogxc, 2022-08-17:
> So, I don't think I've ever tried to estimate a distance from the screen. How often does this situation arises?

I do it often.
(And if it would not be used like this there would anyway not be much sense in having a scale field at all)

lordfolken, 2022-08-17:
I think enroute is implementing it correctly:
[image]

groundhogxc, 2022-08-30:
> I think enroute is implementing it correctly

Well, yes, with the above mentioned drawback that it messes up a lot of map space; which the XCSoar approach avoids.