XCSoar/XCSoar #231: Flarm Range Analysis Dialog

dspreitz, 2019-08-07:
FLARM introduced a new Range Analyser feature in Firmware Version 6.80. Antenna Range is now computed inside the device and a permanent statistics is maintained within the FLARM.

This is the discription of the interface for this feature:
`$PFLAN,A,RANGE,<RangeType>,<Channel>,<Range1>,<Range2>,...<RangeN>`
To request the range info, send a '$PFLAN,R,RANGE' to the device.
The device answers with 20 circle sectors (18° each - top view) in clockwise direction (0° = direction of flight).
Channel is either A or B and denotes the Antenna channel in use. 
Stat denotes how many data samples were used for the analysis.

`$PFLAN,A,RANGE,TIMESPAN,<TimestampStart>,<TimestampEnd>`
Timestamp in UNIX epoch gives the start and end of data acquisition period of given analysis.

Example:
```
> $PFLAN,R,RANGE
$PFLAN,A,RANGE,RFTOP,A,5600,4800,3600,2400,1200,1200,...* $PFLAN,A,RANGE,RFTOP,B,4800,3600,2400,1200,1200,1200,...* $PFLAN,A,RANGE,STATS,5000*
$PFLAN,A,RANGE,TIMESPAN,1562000000,1563000000* 
```
The statistic can be reset (in case of e.g. antenna setup change) by:
```
$PFLAN,A,RANGE*
> $PFLAN,S,RESET
$PFLAN,A,RESET*
```

This feature request proposes to include a GUI feature to display a dialoge showing the FLARM Range in XCSoar (Config --> NMEA Settings --> Flarm --> Properties) together with a possibility to reset the statistics.

rawtaz, 2019-08-07:
Thanks for an informative feature suggestion! Are the measurements in metres, or what's the unit for the circle sectors returned?

dspreitz, 2019-08-07:
Yes, the distance per sector is in meters.

Exadios, 2019-08-07:
What's your source for this info? Is there any mention of a maximum for 'N' - as in 'RangeN'?

dspreitz, 2019-08-07:
The source is the 'DATA PORT INTERFACE CONTROL DOCUMENT (ICD)' -FTP-012 by FLARM.
Can be requested by everybody on their website: https://flarm.com/support/manuals-documents/icd-download/
Maximum for N is 20: "Expect 20 circle sectors of 18 degrees each".

Exadios, 2019-08-07:
Thanks for the source and the clarification. I should have read what you wrote more carefully. :)

lordfolken, 2019-08-09:
I like this, please implemnent and submit a bug report @dspreitz

dspreitz, 2019-08-13:
I am presently entirely clueless how GUI programming in XCS works. Where should I start looking at code to understand how to implement a corresponding FLARM-Range dialog?

lordfolken, 2019-08-13:
The source code is generally organized, like so:
http://download.xcsoar.org/releases/6.8.11/XCSoar-developer-manual.pdf#page=22

So in the `src/Dialogs` you will find all the dialogs. Pick one and deduce from there. 

The modifications for the FLARM protocol should be in `src/Device/Driver/FLARM`

dspreitz, 2026-09-10:
Still current, checked against the latest specification — but the
description above is from 2019 and has since gone out of date in three
ways that matter to anyone implementing it.

## The feature is alive

`$PFLAN` is in the current [Data Port ICD FTD-012 version 7.22,
2025-09-24](https://www.flarm.com/wp-content/uploads/2025/09/FTD-012-Data-Port-Interface-Control-Document-ICD-7.22.pdf),
chapter **8.18**, no longer 8.17 — `PFLAJ` was inserted ahead of it.

It has not been touched in years. Comparing the chapter across 7.19,
7.20 and 7.22 with the layout differences normalised away leaves no
differing line at all, while `PFLAM` and `PFLAP` were added around it
and `PFLAU` was corrected. The firmware release notes are at 7.43
(2026-02) and show active work on the underlying mechanism rather than
a wind-down — "Improved robustness of CARP radio signal recording and
reporting", a fix for CARP data missing from IGC output, and under
"Changes since 7.40": *"PowerFLARM Flex: Introduced InstaCARP with fast
convergence using RSSI"*. Faster convergence makes the statistic useful
after less flight time, which if anything strengthens the case for a
dialog.

One licensing note worth having: with ICD 7.20 the confidentiality
status changed from `NDA` to **`Public`**, and 7.22 keeps it. In 2019 I
wrote that the document could be requested; it can now simply be cited,
which is easier to live with in a GPL project.

## Where the description above is wrong

| This issue (2019, ICD 7.11) | ICD 7.22 |
| --- | --- |
| one statistic, `RFTOP` | **three**: `RFTOP` mean range in metres, `RFCNT` number of data points per sector, `RFDEV` standard deviation |
| `<RangeType>` | `<StatisticType>` |
| "Stat denotes how many data samples were used" | `STATS,<NumberOfPointsTop>` |
| "20 circle sectors" | *"Currently, there are 20 sectors of 18° each, **but this may change in the future**"* |

Version 7.14 (2021-05-10) is where this happened: *"Added new
statistics in PFLAN output, changed semantic of RFTOP."*

Three consequences for an implementation:

1. **Do not hard-code 20 sectors.** The spec reserves the right to
   change the count, so the parser should take however many fields
   arrive and the renderer should divide the circle by that number.
2. **Empty fields are expected.** *"An empty field implies that the
   value could not be reliably computed for the relevant sector."* A
   gap is not zero, and drawing it as zero would misreport a blind
   spot as a measurement.
3. **`RFCNT` gates `RFTOP`.** The spec says outright: *"Each sector
   should be checked to have a sufficiently large RFCNT value to be
   significant."* A sector backed by three data points must not look
   like a range figure. This is also the argument for showing `RFCNT`
   and `RFDEV` in the dialog rather than only the pretty polar plot —
   without them the pilot cannot tell a real null from an unsampled
   direction.

Also: *"Not available on Classic FLARM-based devices"* — PowerFLARM
only, so the entry needs to degrade gracefully.

## Where XCSoar stands

`PFLAN` does not appear anywhere in the tree. The driver handles
`PFLAA`, `PFLAC`, `PFLAE`, `PFLAF`, `PFLAI`, `PFLAJ`, `PFLAM`,
`PFLAQ`, `PFLAR`, `PFLAU`, `PFLAV` and `PFLAX`, but not this one.

The surrounding pieces do exist, which makes this smaller than it
looks:

* `FlarmDevice::RequestSetting()` / `SendSetting()` in
  `src/Device/Driver/FLARM/Settings.cpp` already do request/response
  over `PFLAC,R` and `PFLAC,S`. `PFLAN,R,RANGE` needs the same shape,
  but its answer is several sentences terminated by a bare
  `$PFLAN,A,RANGE*`, so it does not fit `GetSetting()` as it stands.
* `src/Dialogs/Device/ManageFlarmDialog.cpp` is the dialog this issue
  asks to extend — it currently carries Setup, Reboot and Simulation.

One trap for whoever picks this up:
`src/Dialogs/Device/FLARM/RangeConfigWidget.cpp` sounds like the right
file and is not. It configures `RANGE`, `VRANGE`, `PCASRANGE` and
friends — the alarm ranges the device should use, not the range it has
measured.

lordfolken, 2026-09-10:
Refreshed the issue description against FTD-012 ICD 7.22 (thanks @dspreitz).

Still wanted: a PowerFLARM-only range plot on the existing FLARM manage dialog, with reset. Parser first, then the dialog.

lordfolken, 2026-10-18:
The PowerFLARM range dialog from this issue is on master.

- #3289 reads and resets the CARP statistics (`$PFLAN,R,RANGE` / `$PFLAN,S,RESET`).
- #3303 adds the plot on Devices → Manage: antenna A/B, weak and missing sectors, the minimum-range line, and reset.

closed 2026-10-18 by lordfolken