XCSoar/XCSoar #232: panorama mode - center

scls19fr, 2019-08-09:
Hello,

XCSoar is a  very nice software but it could be even better with a button in panorama mode to (re)center map on glider position.

Kind regards

lordfolken, 2019-08-09:
"Pan off" does exactly that. Please don't use the bug tracker for end user questions.

closed 2019-08-09 by lordfolken

scls19fr, 2019-08-10:
No with "Pan off" you quit panorama mode and are back to navigation mode which is not what I'm asking for here.

Autocenter button in panorama mode could center map on glider (in panorama mode) without quitting panorama mode.

I really think that UI interaction should be improved to have a quite similar gesture experience in panorama mode and in navigation mode because current behavior in counter intuitive.

For example, 

in nav mode, 
we use 2 fingers touch and move 2 fingers to pan (so there is support for multi touch which is nice)
1 finger gesture down to up to zoom in
1 finger gesture up to down to zoom out

in panorama mode
1 finger move to pan
zoom in / out can only be done using buttons (up/down and down/up gestures are not available in this mode)

My opinion is that, in panorama mode, pan should be done using 2 fingers touch also.
By doing this it should be possible to zoom in / zoom out using down/up and up/down gesture (like in nav mode)

If panorama mode and navigation mode had zoom in / zoom out and pan similar gestures an autocenter button won't be necessary because "Pan off" could be solution ... but here the main problem is that user need to behave differently for these 2 basic features (zoom/pan) depending on the mode which is currently enabled (panorama or nav) and this is really a problem which shouldn't simply be ignored.