XCSoar/XCSoar #234: Scale is not displayed in panorama mode

scls19fr, 2019-08-09:
Hello,

I'm facing issue with panorama mode... scale is not displayed (see screenshot).
I'm using a Huawei P30 Lite (6.15" - 2312 x 1080 resolution - 415 DPI).

[image]

On a much bigger screen (Acer Chromebook CB5-132T-C8VM with 11" screen) scale is displayed correctly.

[image]

Scale should never be hidden by buttons.

Kind regards

lordfolken, 2019-08-09:
Yes, please write a bugfix.

scls19fr, 2019-08-10:
I currently haven't dive into XCSoar code previously but I have some Java / Kotlin skills... 
So could you point me where is the code which manage scale feature?

lordfolken, 2019-08-10:
have a look into `src/Renderer/MapScaleRenderer.cpp`

scls19fr, 2019-08-10:
I'm wondering what are the reasons of this `3` factor in

https://github.com/XCSoar/XCSoar/blob/b2ae619d507419cd92f0fb629dc078a5b36be9fe/src/Renderer/MapScaleRenderer.cpp#L51

I have been able to reproduce this issue on my Macbook, so I wanted to use it as a testing environment... but wasn't able to compile XCSoar.

```
$ make
```

raises

```
configure: error: You will need to substantially rewrite libxtiff to
build libgeotiff without libtiff
Traceback (most recent call last):
  File "./build/thirdparty.py", line 140, in <module>
    x.build(toolchain)
  File "/Users/scls/github/scls19fr/XCSoar/build/python/build/autotools.py", line 99, in build
    build = self.configure(toolchain)
  File "/Users/scls/github/scls19fr/XCSoar/build/python/build/autotools.py", line 86, in configure
    subprocess.check_call(configure, cwd=build, env=toolchain.env)
  File "/Users/scls/anaconda3/lib/python3.7/subprocess.py", line 347, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/Users/scls/github/scls19fr/XCSoar/output/src/libgeotiff-1.4.2/configure', 'CC=clang', 'CXX=clang++', 'CFLAGS=-Os -g -ffunction-sections -fdata-sections -fvisibility=hidden  -mmacosx-version-min=10.7 -target x86_64-apple-darwin', 'CXXFLAGS=-Os -g -ffunction-sections -fdata-sections -fvisibility=hidden  -mmacosx-version-min=10.7 -target x86_64-apple-darwin', 'CPPFLAGS=-isystem /Users/scls/github/scls19fr/XCSoar/output/OSX64/lib/x86_64-apple-darwin/root/include -DNDEBUG -I./output/OSX64/include -DHAVE_POSIX -DHAVE_STDINT_H -DHAVE_UNISTD_H -DHAVE_VASPRINTF -isystem ./output/OSX64/lib/x86_64-apple-darwin/root/include -DEYE_CANDY -DTESTING -isystem /include -D__ASSERT_MACROS_DEFINE_VERSIONS_WITHOUT_UNDERSCORES=0  ', 'LDFLAGS=-L/Users/scls/github/scls19fr/XCSoar/output/OSX64/lib/x86_64-apple-darwin/root/lib -stdlib=libc++ ', 'LIBS= -lz', 'AR=ar', 'ARFLAGS=rcs', 'RANLIB=ranlib', 'STRIP=strip', '--host=x86_64-apple-darwin', '--prefix=/Users/scls/github/scls19fr/XCSoar/output/OSX64/lib/x86_64-apple-darwin/root', '--enable-silent-rules', '--disable-shared', '--enable-static', '--disable-doxygen-doc', '--disable-doxygen-dot', '--disable-doxygen-man', '--disable-doxygen-html']' returned non-zero exit status 1.
make: *** [output/OSX64/lib/x86_64-apple-darwin/stamp] Error 1
```

I will try to compile on a Linux machine when possible (maybe next week) if no workaround.

lordfolken, 2019-08-10:
The build target is debian 9 (stretch). There is a vagrant box in the sources: https://github.com/XCSoar/XCSoar/tree/master/ide/vagrant

Or i have docker container, that might need some work: https://github.com/lordfolken/xcsoar-docker-build

for a plain installation the dependencies are in: https://github.com/XCSoar/XCSoar/tree/master/ide/provisioning

CptFrikadel, 2019-11-12:
I think I fixed it.. See Pull request

lordfolken, 2021-04-08:
I believe we can close this issue.

closed 2021-04-08 by lordfolken

scls19fr, 2021-04-08:
Thanks @CptFrikadel for fixing this.
Related to map scale... fixing #230 will be a great improvement... but it seems to be an harder task and that there's not currently a general consensus