XCSoar/XCSoar #235: Sim mode - define initial position

scls19fr, 2019-08-09:
Hello,

Reading sim mode it's unclear to me how to define initial position (using GPS position or using a waypoint).
Is there a config file to modify? (to set initial position, but also maybe to set some others constants such as max cruise speed in sim mode ...)

Kind regards

lordfolken, 2019-08-09:
This is an instrument, not a smartphone app. It requires study and understanding to be used. Its not intended to be easy or self explanatory. 

All these questions are answered in the XCSoar in a flash guide. http://download.xcsoar.org/releases/6.8.11/XCSoar-in-a-flash.pdf

closed 2019-08-09 by lordfolken

scls19fr, 2019-08-10:
Dear @lordfolken, I'm aware of existence of this manual (and have ever read it previously).

> p3 
> Run XCSoar for the first time
> Run XCSoar. With startup, XCSoar asks you to enter either a real life (Fly) or a simulation
> (Sim) mode. In Sim mode, you can input speed, heading, and height manually in order to do
> some exercising on the ground. 
> 

even the complete manual http://download.xcsoar.org/releases/6.8.11/XCSoar-manual.pdf

> p22
> XCSoar comes with a simple interface allowing for conducting a
> simulated flight. Depending on the hardware platform, there are
> different methods for altering values of bearing, speed, an height.
> Simulation is intended for your first familiarization with XCSoar in
> action. If you like the idea simulating a more realistic thing at
> home, you should acquire a ”true” soaring simulator, to be connected to XCSoar.
> On the map screen, clicking/touching the glider symbol with touchscreen or mouse and dragging causes the glider to move in the
> direction of the drag, the speed being proportional to the length of
> the drag.
> With buttons, the aircraft speed, altitude and direction may be
> changed using the Infoboxes. The following may not be available
> in total on every hardware platform, but on any of the platforms
> XCSoar supports, a full set of inputs needed for simulation purposes is possible.
> By pushing an Infobox, you select a value to be altered by buttons, either hardware or menu buttons. The aircraft altitude can
> be adjusted by selecting the GPS altitude InfoBox Alt GPS , and
> touching the up or down key or buttons on the touchscreen. The
> airspeed can be adjusted by selecting the Infobox ground speed
> V Gnd , touching up or down key or menu buttons. The glider’s
> track can be adjusted by selecting the track InfoBox Track , and
> touching up/down buttons.
> With either of the Alt GPS or V Gnd selected, the glider’s direction may be changed using the left/right keys.
> Other controls, buttons and menus work the same as in FLY mode.

I don't see any information about how to set initial position in sim mode (in this paragraph).

But my opinion is that, as XCSoar is an instrument and not a smartphone which have a simulation mode for learning purpose (before using it in real flight), the description in manual of this sim mode should be improved.

After doing some experiments I noticed that initial position in sim mode can be set using task actions menu "Set as new home".

scls19fr, 2019-08-10:
PR https://github.com/XCSoar/XCSoar/pull/237 for improving doc about this point have just been sent. Please review.

PR #237 (2019-08-10): Update XCSoar-in-a-flash.tex
Closes #235
commit cfb485e0 (2019-08-10): Update XCSoar-in-a-flash.tex

Closes #235