XCSoar/XCSoar #238: Gesture personalisation / mapping

brunotag, 2019-08-11:
Hi all, just wondering if there is appetite for a gesture mapping system / menu, that would allow to map gestures to actions/commands.
Such commands might (or not) be mapped to different gestures on different Info Pages or Info Modes.

To clarify, I am not proposing to switch to a different "gestures model", or multitouch, but rather allow the users to map different gestures to different commands/actions (on different Pages / Modes if feasible).

At this stage, I'd be just keen investigating how many other people would be interested, so add a 👍 in case.

lordfolken, 2019-08-12:
They are mappable. Have a look at https://github.com/XCSoar/XCSoar/blob/master/Data/Input/default.xci type=gesture. You can for example map some different action on UP/Down.

Unfortunately the manual isn't that clear about how to extend this. But in theory you can match any gesture to any action that is addressable. Gesture is just like any button press. 

Please consider extending the xcsoar-manual,  specifically section `3.12 Intefacing with Gestures` and `14.12 Input Events`. It also references the developer manual, section 8.1 Input Events, but gestures are not mentioned there.

brunotag, 2019-08-12:
Thank you!

Wow you have a developer manual! That's much better than trying to figure out stuff by reading code :o)

I should have read it before .. instead I've looked at the code (superficially) before posting.. and finding stuff like [this ](https://github.com/XCSoar/XCSoar/blob/b120ea7bb14eea14c3e73d77a9f23f3ba886876e/src/Dialogs/dlgAnalysis.cpp#L612) made me think the gestures are tightly coupled to commands.
Maybe that's dead code?

So then [here](https://github.com/XCSoar/XCSoar/blob/9041dc620a76ce72906021463f42ce58f0353425/src/Input/InputParser.cpp#L137), or [here](https://github.com/XCSoar/XCSoar/blob/6412cdf373f59f627f78fd8751c098ae7d45ea91/src/Input/InputDefaults.cpp#L108) has to be where you load the config.. and [here](https://github.com/XCSoar/XCSoar/blob/cf33335b3412b06eaaed2e14214af7927d6c0828/src/Input/InputEvents.cpp#L345) and [here](https://github.com/XCSoar/XCSoar/blob/cf33335b3412b06eaaed2e14214af7927d6c0828/src/Input/InputEvents.cpp#L357) must be were you use them.

I can't find that file in the XCSoarData in my Android phone... and I can't figure out if I should I create it and where? I can't find the paths mentioned in the manual.
Actually I'm not 100% sure I can do this from the app.. the manual makes it sound like it is possible, but it is also done by generating code at compile time..?

brunotag, 2019-08-13:
Thanks.
Sorry, I really am trying not to be difficult, but I don't understand what "XCSoar Menu/Settings/Input File" is supposed to be or to mean.
Would that be a menu? I found it mentioned in the dev manual, yes, and on some web pages online, but there is no "Settings" menu that I can find, at least in the 6.8 Android version I'm running.
And I can't find "Input Files" in any of the menus, not even in "Expert" mode.

lordfolken, 2019-08-13:
Sorry i copied and pasted from the manual, and its incorrect: (please send a bugfix for the documentation) 
1. double tab on main screen
2. Config
3. system
4. look
5. language, input
6. Events -> Choose file

brunotag, 2019-08-13:
Oh and I thought I tried everything.. thanks.

Yeah I'm trying to build the ./docs files, I understand the PDF are generated from there? But I haven't had much success up to now. I'm using MikTex and the batch file on windows, but there must be some missing prerequisite because it cries for missing files.

How do you build the docs? Do you use that option, or doxygen, or something else?

lordfolken, 2019-08-13:
Windows is at this point the vt320 for the 21st. century. Just install virtualbox and vagrant from https://chocolatey.org Then take the vagrant file in ide/vagrant/.  This will install a vm that is able to compile / generate the docs.

brunotag, 2019-08-14:
Right... Vagrant is indeed mentioned in the dev manual, just not in the published one yet.
Vagrant is awesome.

btw, docs say 

> The Input System is deprecated! It is being replaced by a Lua
> scripting engine.

Is it still the case? Even for gestures?
Lua is great and powerful, but maybe a bit overkill for simple customisations, compared with a simple mapping file.

lordfolken, 2019-08-14:
Its correct that this is on the way out, and going to be replaced with  Lua script(s). Max has been working at this for some time now and its slanted for the 7.0 release. 
Ultimately it will allow you to trigger functions, and several actions not just a single event/reaction as is the case with the current system.
There is a lua section in the manual. I haven't tried it yet.

brunotag, 2019-08-18:
I see.. well maybe not even worthwhile updating the docs if it'll be deprecated.
Anyway, for what it's worth, here it is https://github.com/XCSoar/XCSoar/pull/243

closed 2020-05-04 by lordfolken