XCSoar/XCSoar #242: Describe Waypoint_Details file format in XCSoar User Manual

lordfolken, 2019-08-14:
The documentation does not describe the waypoint_details file anywhere. 

The Format is currently: 
[WAYPOINT_NAME] <- must match waypoint in cup
Text Text
Multiline <= appears in the description field and as long text on the following waypoint page
file=/relative/url/to/xcsoardata=*.jpg <= clickable file to open with default associated picture viewer
file=/relative/url/to/xcsoardata=*.pdf <= clickable file to open with defualt associated image viewer

Buzz33, 2021-06-23:
Try to update the manual relating to the the waypoint details - during the next weeks.

Buzz33, 2021-07-22:
By using Github desktop I finally updated the tex files of Ch4 Tasks, CH Installation, Ch11 Configuration and CH12 data files - in German language. I also described "displaying pictures" in XCSoar, using an example with some additional screenshots whitch I uploaded to "figures" . It is in my forked repository by now, pushed to master. 

What is to do next? Maybe some advise would be useful, before I mess up anything  - because I am perfectly new to github and tex. ;-)

closed 2022-05-20 by lordfolken