XCSoar/XCSoar #244: CUP file is not displayed correctly

scls19fr, 2019-08-21:
Hello,

I wrote a small Python script to convert a CSV file with degree/min/sec coordinates of waypoints to CUP file.

```bash
$ cat poitiers_vfr.cup
name,latitude,longitude,desc
E,46°33’32”N,000°30’02”E,"St Julien - l’Ars : voie ferrée"
EA,46°35’30”N,000°23’40”E,"Usine Buxerolles (cheminée blanche, route)"
W,46°38’48”N,000°10’13”E,"Vouillé/route, rivière Auxance"
WA,46°37’43”N,000°15’51”E,"Rond-point de Moulinet"
```

and `convert_wpt_cup.py` script

```python
"""
Convert CSV file of waypoints with degrees/minutes/seconds to .cup files

Usage:

$ python convert_wpt_cup.py poitiers_vfr.csv --name "%s@LFBI" --country "FR"
"""

import click
import re
import os
import pandas as pd
from aerofiles.seeyou.writer import Writer
from aerofiles.seeyou.converter import WaypointStyle

COORD_PATTERN = re.compile(r"(\d+)°(\d+)['’](\d+)[\"”]([NSEW])")
SGN_DIR = {"N": 1, "S": -1, "E": 1, "W": -1}

def convert_dms(s):
    m = COORD_PATTERN.match(s)
    (dd, mm, ss, d) = m.groups()
    dd = int(dd)
    mm = int(mm)
    ss = int(mm)
    sgn = SGN_DIR[d]
    return sgn * (dd + mm / 60.0 + ss / 3600.0)

@click.command()
@click.argument("csv_input")
@click.option("--name", default="", help="Name of waypoint")
@click.option("--country", default="FR", help="Country code")
def main(csv_input, name, country):
    print("Loading %s" % csv_input)
    df = pd.read_csv(csv_input)
    print(df)
    df["shortname"] = df.name.copy()
    df["country"] = country
    if name != "":
        df.name = df.name.map(lambda s: name % s)
    assert df.name.is_unique, "names must be unique"
    df.latitude = df.latitude.map(convert_dms)
    df.longitude = df.longitude.map(convert_dms)
    print(df)
    (root, _) = os.path.splitext(csv_input)
    cup_output = root + ".cup"
    print("Writing %s" % cup_output)
    with open(cup_output, "wb") as fd:
        w = Writer(fd)
        for (_, wpt) in df.iterrows():
            w.write_waypoint(
                wpt["name"],
                wpt.shortname,
                wpt.country,
                wpt.latitude,
                wpt.longitude,
                elevation=u"",
                style=WaypointStyle.NORMAL,
                runway_direction=u"",
                runway_length=u"",
                frequency=u"",
                description=wpt.desc,
            )

if __name__ == "__main__":
    main()
```

This script is using Python Pandas and [aerofiles](https://github.com/Turbo87/aerofiles) from @Turbo87 

using this script like

```bash
$ python convert_wpt_cup.py poitiers_vfr.csv --name "%s@LFBI" --country "FR"
Loading poitiers_vfr.csv
  name    latitude    longitude                                        desc
0    E  46°33’32”N  000°30’02”E             St Julien - l’Ars : voie ferrée
1   EA  46°35’30”N  000°23’40”E  Usine Buxerolles (cheminée blanche, route)
2    W  46°38’48”N  000°10’13”E              Vouillé/route, rivière Auxance
3   WA  46°37’43”N  000°15’51”E                      Rond-point de Moulinet
4    S  46°26'55"N  000°16'10"E                             Réservoir d'eau
      name   latitude  longitude                                        desc shortname country
0   E@LFBI  46.559167   0.508333             St Julien - l’Ars : voie ferrée         E      FR
1  EA@LFBI  46.593056   0.389722  Usine Buxerolles (cheminée blanche, route)        EA      FR
2   W@LFBI  46.643889   0.169444              Vouillé/route, rivière Auxance         W      FR
3  WA@LFBI  46.626944   0.254167                      Rond-point de Moulinet        WA      FR
4   S@LFBI  46.440556   0.271111                             Réservoir d'eau         S      FR
Writing poitiers_vfr.cup
```

I'm getting the following cup file

```bash
$ cat poitiers_vfr.cup
name,code,country,lat,lon,elev,style,rwdir,rwlen,freq,desc
"E@LFBI","E",FR,4633.550N,00030.500E,,1,,,,"St Julien - l’Ars : voie ferrée"
"EA@LFBI","EA",FR,4635.583N,00023.383E,,1,,,,"Usine Buxerolles (cheminée blanche, route)"
"W@LFBI","W",FR,4638.633N,00010.167E,,1,,,,"Vouillé/route, rivière Auxance"
"WA@LFBI","WA",FR,4637.617N,00015.250E,,1,,,,"Rond-point de Moulinet"
"S@LFBI","S",FR,4626.433N,00016.267E,,1,,,,"Réservoir d'eau"
```

this cup file is rendered correctly in GPXSee (thanks to https://github.com/tumic0/GPXSee/issues/219 ) but doesn't seems to be readable by XCSoar (waypoint are not rendered)

I don't understand what's wrong.

Is my script wrong? (it may be possible but why is it rendered correctly in GPXSee and not in XCSoar.

Some help will be great.

Kind regards

kerel-fs, 2019-08-23:
The elevation field is empty for all waypoints in your example file.

XCSoar Manual - Section 14.3 Waypoints says:
> If the elevation of any waypoints is set to zero in the waypoint file,
then XCSoar estimates the waypoint elevation from the terrain
database if available.

Otherwise the waypoint is ingored.
While using the map database "FRA_FULL_HighRes.xcm" the waypoints in your cup file are rendered successfully with XCSoar (master & latest release).

scls19fr, 2019-08-24:
Thanks @kerel-fs but I only see in doc information about waypoint with elevation set to zero... not about waypoint with empty elevation.
empty and zero doesn't mean the same thing... so doc should be more explicit. 
moreover waypoints should't be silently be ignored... user should be informed about that.

kerel-fs, 2019-09-15:
@scls19fr: sorry, I missed your response until now.

Indeed, the documentation could be improved here! Can you submit a patch for this?

Note that waypoints are only ignored if there is no map data available at the specific location, so this shouldn't occur during normal usage. Nevertheless, notifying the user about ignored waypoints would be a nice feature (but with a lower priority I guess).

scls19fr, 2019-09-16:
I'm not sure to understand what you are saying here.

From my experience, I noticed that with such waypoints (with empty elevation field), waypoints are simply ignored despite they are located in a place which have map data available.

kerel-fs, 2019-09-16:
I just tested again and was able to confirm that XCSoar doesn't ignore waypoints with empty elevation field when map data is available. Please let me know if you were able to find an example for the behavior you experienced, maybe there is another bug.

Tested with the following waypoint file (using `FRA_FULLHighRes.xcm`):
```
name,code,country,lat,lon,elev,style,rwdir,rwlen,freq,desc
"E@LFBI","E",FR,4633.550N,00030.500E,,1,,,,"St Julien - l’Ars : voie ferrée"
"EA@LFBI","EA",FR,4635.583N,00023.383E,,1,,,,"Usine Buxerolles (cheminée blanche, route)"
"W@LFBI","W",FR,4638.633N,00010.167E,,1,,,,"Vouillé/route, rivière Auxance"
"WA@LFBI","WA",FR,4637.617N,00015.250E,,1,,,,"Rond-point de Moulinet"
"S@LFBI","S",FR,4626.433N,00016.267E,,1,,,,"Réservoir d'eau"
```

See also the following code:
https://github.com/XCSoar/XCSoar/blob/2a5719536f488a86ef49d87a113c71bdf5ed26a2/src/Waypoint/WaypointReaderSeeYou.cpp#L244-L249

scls19fr, 2019-09-17:
You are right @kerel-fs 
It works fine. My problem that when editing cup file, my Android text editor have behaved incorrectly with CRLF (replacing with only LF). After fixing this it works fine. Sorry.

closed 2019-09-17 by scls19fr