XCSoar/XCSoar #255: Always upload latest XCSoar dev manual via CI

kerel-fs, 2019-09-25:
The gitlab-CI can be used to generate & upload the latest XCSoar dev manual for the master branch to xcsoar.org

elgandoz, 2020-09-28:
This has been implemented and it used to work, but it stopped working since 03/07/2020 (see http://download.xcsoar.org/nightly_builds/)

I post this here because the existing issue #358 was closed.

lordfolken, 2020-09-28:
@elgandoz the job was disabled, because it filled the disk of the server rather reliably, tearing down the forum. 
I have a new approach with github-actions in the works: 
https://github.com/lordfolken/XCSoar/tree/dockerupdates

elgandoz, 2021-02-21:
Hi all. Is there any update on this issue? I think it would help a lot the testing having an easy way to get the latest APK (or other executables). If this is not really moving, is there a chance to restore the old task, maybe with a basic cleanup removing all the files but the executables, in order to save space/resources? 
Another alternative, for Android,  could be publishing the beta with a shorter cycle. Thanks

lordfolken, 2021-06-09:
we do have automation now to generate the manual. There is also a release automation. Needs some uploading mechanism.

elgandoz, 2021-06-09:
Thanks @lordfolken, this is great news. I also noticed the publishing on the store happens much more frequently.
Having a nightly-build .apk would still help to try features and catch early bugs.

closed 2021-07-25 by lordfolken