XCSoar/XCSoar #259: Optimization issue

SBollatti, 2019-09-27:
A few of us (Argentinian) flying AAT in Condor Soaring, we got this issue and its consequence not able to get the real task Speed.
[image]
[image]
[image]
[image]
[image]

igc's link: [https://www.condor-club.eu/besttimes/0/?id=12977](url)

My extremely huge regards to this community and so thankful to use XCSoar.
Seba
Federación Argentina de Vuelo a Vela Virtual

lordfolken, 2019-10-05:
Could you describe the issue a bit?
 - What do you expect to happen?
 - What did happen to your observation?

Please also attach the task file.

SBollatti, 2019-11-02:
Attached a today new flight
[image]

kerel-fs, 2019-11-03:
The task you uploaded is located in Slovenija whereas your IGC file belongs to a flight in Argentina.
Can you provide the task which matches the flight as well, please?

Some wild guessing, as you didn't describe the issue more precisely yet:
The "optimization issue" you refer to can be seen in the last example around the TP4 area where the target point used for task length calculation doesn't optimize the total length of the flown path.
These target points can be moved manually inside the AAT areas though. Maybe this happened earlier in the flight?

SBollatti, 2019-11-03:
Attached

<Task fai_finish="0" finish_min_height_ref="MSL" finish_min_height="0" start_max_height_ref="MSL" start_max_height="1500" start_max_speed="0" start_requires_arm="1" aat_min_time="7200" type="AAT">
	<Point type="Start">
		<Waypoint altitude="50" comment="Alt. limits=0-1500" id="0" name="Partida A[0-15]">
			<Location latitude="-34.4433" longitude="-59.4373"/>
		</Waypoint>
		<ObservationZone length="10000" type="Line"/>
	</Point>
	<Point type="Area">
		<Waypoint altitude="45" comment="" id="2" name="Suipacha">
			<Location latitude="-34.7717" longitude="-59.6731"/>
		</Waypoint>
		<ObservationZone radius="10000" type="Cylinder"/>
	</Point>
	<Point type="Area">
		<Waypoint altitude="65" comment="" id="0" name="Los angeles">
			<Location latitude="-34.457" longitude="-60.18"/>
		</Waypoint>
		<ObservationZone radius="25000" type="Cylinder"/>
	</Point>
	<Point type="Area">
		<Waypoint altitude="44" comment="" id="4" name="Cap Sarmiento Ae">
			<Location latitude="-34.1639" longitude="-59.7311"/>
		</Waypoint>
		<ObservationZone radius="15000" type="Cylinder"/>
	</Point>
	<Point type="Finish">
		<Waypoint altitude="46" comment="" id="1" name="Albatroz">
			<Location latitude="-34.4261" longitude="-59.4573"/>
		</Waypoint>
		<ObservationZone radius="3000" type="Cylinder"/>
	</Point>
</Task>

AATMinTime="7200"
AATTimeMargin="180"
AbortTaskMode="1"
AircraftSymbol="2"
AppAveNeedle="1"
AppDialogTabStyle="1"
AppGaugeVarioAvgText="1"
AppGaugeVarioBallast="1"
AppGaugeVarioBugs="1"
AppGaugeVarioGross="0"
AppGaugeVarioMc="1"
AppGaugeVarioSpeedToFly="0"
AppInfoBoxBorder="2"
AppInverseInfoBox="1"
AppScaleRunwayLength="1"
AppStatusMessageAlignment="0"
AudioVario2="0"
AutoCloseFlarmDialog="1"
AutoMc="0"
AutoMcMode="2"
AutoWind="3"
AutoZoom="0"
AverEffTime="0"
BlockSpeedToFly="1"
ClimbMapScale="97"
CloudEnabled="0"
CruiseMapScale="648"
DetourCostMarker="1"
DeviceA="Condor"
DeviceB=""
DisplayOrientation="1"
DisplayText="6"
DisplayTrackBearing="1"
DrawTerrain="1"
DrawTopology="1"
[... 242 lines cut ...]
RoutePlannerAllowClimb="1"
RoutePlannerMode="3"
RoutePlannerUseCeiling="1"
SafetyAltitudeArrival="0.000000"
SafetyAltitudeTerrain="0.000000"
SafetyMacCready="0"
SlopeShadingType="3"
SnailTrail="3"
SnailType="1"
StartRadius="5000.000000"
StartType="1"
TaskType="5"
TeamcodeRefWaypoint="-1"
TerrainBrightness="36"
TerrainContours="1"
TerrainContrast="151"
TerrainRamp="6"
TurningReach="2"
UIScale="100"
UserLevel="1"
VarioDeadBandEnabled="0"
WaveAssistant="1"
WayPointLabelSelection="1"
WaypointArrivalHeightDisplay="3"
WindArrowStyle="1"

kerel-fs, 2019-11-05:
I was able to reproduce the issue with XCSoar version 6.8.12-Android.

 **Steps to reproduce**
- Load the given profile & task, start the replay using the provided IGC file
  (I had to remove the start hight limit to get the same start point as you, but this shouldn't matter here, additionally I disabled "Arm start manually")
- At each turn point: Press "Arm turn" to advance to the next turnpoint
- After pressing "Arm turn" at TP4 the optimizer moved the fourth target point at the non-optimal position as shown above

- The manual movement of previous target points is not possible, so in order to get an approximately correct optimized task distance, let the replay advance until the finish cylinder is reached. Then pause the replay and press "previous turnpoint". Now it's possible to move the target point in the TP4 area to the approximately correct location (optimizing the flight distance) manually again.

The resulting task distance is ~263 km (as given in the "Status: Task" view) contrary to 254 km when using the built-in optimization.

MaxKellermann, 2021-07-04:
Phew. Analyzing this bug took this whole sunday. In the end, it was simply caused by the wrong sort order in the convex hull algorithm, and I'm wondering if there's more fallout due to this 12 years old bug.

closed 2021-07-04 by MaxKellermann