XCSoar/XCSoar #276: Maximum height integration (for Paramotor)

DiegoTheWolf, 2019-10-11:
Hello,

i did not find anything matching in here so i'm opening this request/idea/issue.

The Maximum height integration is useful for paramotor pilots or everyone who flys with motor and without transponder. In some countries (like Germany) you are only allowed to fly up to 5000 ft MSL or 3500 ft AGL (higher counts) without a transponder.

Now i would like to have implemented:
- altitude warning (like the airspace warning, with prediction)
- in the cross section view, show the flyable band and reduce maximum altitude.

Thanks for looking into it.

Diego
PS: If someone could introduce me to the project i may be able to contribute, since i have some programming (C#/C++/Java) knowledge.

lordfolken, 2019-11-28:
There is no "dedicated" xcsoar team that can implement your request. So you will have to champion and implement this request yourself. 

We have a developer manual that should get you started:
[Developer Manual](https://download.xcsoar.org/releases/6.8.11/XCSoar-developer-manual.pdf)

fb, 2019-12-10:
@DiegoTheWolf, I think you can get what you need using a custom airspace file.

- usable with current and old versions of XCSoar
- no new configuration options
- shown in "closest airspace vertical/horizontal" infoboxes
- shown in in airspace cross-section
- warnings like for any other airspace

I looked up some information about the OpenAir format for airspaces here: http://www.winpilot.com/UsersGuide/UserAirspace.asp and came up with this:
 
These are two circles. You will have to adjust the coordinates and radius to something like *middle of germany* and *500 nm*.

```
AC P
AN Prohibited bove  5000 ft MSL
AH FL100
AL 5000ft MSL
V X=39:29.9 N 119:46.1 W
DC 5

AC P
AN Prohibited above 3500 ft AGL
AH FL100
AL 3500ft AGL
V X=39:29.9 N 119:46.1 W
DC 5
```

fb, 2020-04-26:
No feedback. Closing.

closed 2020-04-26 by fb