XCSoar/XCSoar #287: Screen not fully used on Pixel 4XL

johnthomas22, 2019-11-11:
If someone can point me in the direction of where screen layouts are defined, I could try and work out some suggestions for fixes to save you real developers some time...

klujek, 2019-11-21:
I have similar problem with Xperia 1 with 21:9 screen. I found some info but I think this is related to install files or even programming...

https://android-developers.googleblog.com/2017/03/update-your-app-to-take-advantage-of.html

lordfolken, 2019-11-28:
The link is interesting: Would someone make a pull request with:
```xml
<meta-data android:name="android.max_aspect"
    android:value="ratio_float"/>
```
in android/AndroidManifest.xml

johnthomas22, 2019-11-28:
Yes, the link does seem to point to something that fits the problem.

I've been trying to get a build environment going on a raspberry pi 4 to
see if I can test.

No luck so far, down to no errors, but can't see how to get an APK.

Regards

John

klujek, 2020-02-18:
https://github.com/XCSoar/XCSoar/pull/346

lordfolken, 2020-02-20:
https://github.com/XCSoar/XCSoar/commit/fb8370c2ed5df0e69a1055e3d53d922847ae2ea4 implements this.

closed 2020-02-20 by lordfolken