XCSoar/XCSoar #288: logbook

adrianoellero, 2019-11-11:
what about to add an automatic logbook feature to fill-up EASA FCL.050 logbook?
I am willing to contribute on software development...

lordfolken, 2019-11-18:
XCSoar lives from its contributors. If you want a feature you have to be willing to implement it yourself.

First you need a dev environment. You can either use [vagrant](https://www.vagrantup.com/) with our vagrant file in `ide/vagrant`, or use our [Dockerfile](https://github.com/XCSoar/XCSoar/tree/master/ide/docker) 

For how to to do contributions, please have a look at  the dev [manual](https://download.xcsoar.org/releases/6.8.11/XCSoar-developer-manual.pdf)

There is already a logbook feature, maybe that one can be extended to support EASA FCL.050. Btw a description to what that format entails would also be great.

lordfolken, 2020-05-04:
Are you still interested in contributing this? If so please reopen the issue, or even better provide a pr even in draft.

closed 2020-05-04 by lordfolken