XCSoar/XCSoar #292: Vagrant build fails

dernatsch, 2019-11-20:
When compiling with `make TARGET=ANDROID -j` on the Vagrant box the build fails with:

> /bin/sh: 1: pkg-config not found

dernatsch, 2019-11-20:
I submitted pull request #293 that adds pkg-config to the provisioning script

kerel-fs, 2019-11-20:
There are multiple uses of the ide/provisioning/* scripts:
- the [host system dev environment setup](https://github.com/XCSoar/XCSoar/blob/1aacd42b60e90a946a1028a3a019a25010386a55/doc/manual/en/developsetuplinux.tex#L30-L40)
- the [Vagrant dev environment](https://github.com/XCSoar/XCSoar/blob/master/ide/vagrant/Vagrantfile)
- the [Docker dev environment](https://github.com/XCSoar/XCSoar/tree/master/ide/docker) (which is the base for the [gitlab-ci](https://github.com/XCSoar/XCSoar/blob/master/.gitlab-ci.yml))

Unfortunately I didn't have the Vagrant env in mind when [updating](https://github.com/XCSoar/XCSoar/commits/master/ide/provisioning) the scripts to be run on Debian 10/buster.

Can you try and change the Vagrant setup to be also based on Debian 10/buster? This should also fix the missing dependency.

Note that OpenJDK-8 is required for Android but it's not available in buster, so the Vagrantfile should add the Debian 9/jessie repository to the sources in order to be able to install OpenJDK-8 from there (see [ide/docker/Dockerfile#L2](https://github.com/XCSoar/XCSoar/blob/master/ide/docker/Dockerfile#L2) / https://github.com/XCSoar/XCSoar/pull/284 for the respective line in the docker setup).

closed 2020-05-03 by lordfolken

kxmm, 2020-05-10:
The build/download.py has to be converted using : dos2unix build/download.py

Afterwards the following Error ist thrown:

  UNTAR   boost_1_73_0.tar.bz2
 does not existost-src/lib/boost/patches/no_locale.diff
Applying patch /xcsoar-host-src/lib/boost/patches/no_locale.diff
make: *** [build/libboost.mk:21: output/src/stamp-boost_1_73_0] Error 1

lordfolken, 2020-05-11:
The file build/download.py is in the correct format
file build/download.py 
build/download.py: Python script, ASCII text executable

Can you make clean your sources and be sure that they match the current head (git pull, git status)?

kxmm, 2020-05-11:
I have cloned the project:
Vagrant up 
=>does not mount xcsoar-host-src
->vagrant plugin install vagrant-vbguest
->vagrant vbguest

#sudo make
=>/usr/bin/env: 'python3\r': No such file or directory
 -> apt-get install dos2unix
 ->dos2unix build/download.py
#sudo make
=>/bin/sh: 1: quilt: not found
->sudo apt-get install quilt

#sudo make
  UNTAR   boost_1_73_0.tar.bz2
 does not existost-src/lib/boost/patches/no_locale.diff
Applying patch /xcsoar-host-src/lib/boost/patches/no_locale.diff
make: *** [build/libboost.mk:21: output/src/stamp-boost_1_73_0] Error 1

lordfolken, 2020-05-11:
Your git repo is not recurisvly cloned and is missing the submodules:
git submodule init
git submodule update

your box is up but not provisioned. Either do a vagrant destroy and vagrant up again or vagrant up --provision

kxmm, 2020-05-14:
Thanx. I had to get a new clone t get it work.

I had to :
sudo apt-get install libglm-dev

to avoid the error:  
CXX     output/UNIX/dbg/src/Renderer/GlassRenderer.o
In file included from ./src/Screen/OpenGL/Scissor.hpp:28,
                 from src/Renderer/GlassRenderer.cpp:29:
./src/Screen/OpenGL/Globals.hpp:37:10: fatal error: glm/fwd.hpp: No such file or directory
 #include <glm/fwd.hpp>
          ^~~~~~~~~~~~~
compilation terminated.

would be great if you could add this lib.

kxmm, 2020-06-01:
Hello Philipp

During provisioning (android-NDK) it stops with an error. 

It seems that there is an Java11 Bug with the SDK Manager

 

So I did the following:

 

sudo apt-get install default-jdk-headless vorbis-tools adb

Download commandlinetools from https://developer.android.com/studio?hl=en-419#downloads 

sudo chmod -R a+w [path/to/android/sdk]

 

~/opt/android-sdk-linux/cmdline-tools/latest/bin$ ./sdkmanager "build-tools;28.0.3" "platforms;android-22"

sudo apt-get install libglm-dev

 

Unfortunately I am not able to build the Android Target (WIN64 works fine):

 

 

vagrant@debian10:~/xcsoar-host-src$ make TARGET=ANDROID7

  CXX     output/ANDROID7/dbg/src/Renderer/GlassRenderer.o

In file included from src/Renderer/GlassRenderer.cpp:29:

In file included from ./src/Screen/OpenGL/Scissor.hpp:28:

./src/Screen/OpenGL/Globals.hpp:37:10: fatal error: 'glm/fwd.hpp' file not found

#include <glm/fwd.hpp>

         ^~~~~~~~~~~~~

1 error generated.

make: *** [build/compile.mk:108: output/ANDROID7/dbg/src/Renderer/GlassRenderer.o] Error 1

 

 

Do you have any suggestions?

 

Michael

lordfolken, 2020-06-01:
You have to recursively checkout xcsoar source.  clone with --recurse or do a submodules init and update.

kxmm, 2020-06-01:
I have downloaded the zip from github.
So this should be complete?
Michael
 

Dr.Michael Maurer
Sent from Mobile

lordfolken, 2020-06-02:
We depend on other projects, such as glm and ioio which come in source form.
They are configured as git submodules. 
folken@WOPH:~/git-xcsoar/XCSoar$ git submodule status
+c817e23975b8b73d2251370023164d0556c63019 android/ioio (App-IOIO0504-5-gc817e23)
 416fa93e42f8fe1d85a93888a113fecd79e01453 lib/glm (0.9.5.3-2169-g416fa93e)

They are not part of the git tree of xcsoar.

Some General information:
https://github.blog/2016-02-01-working-with-submodules/

please read
2.1 Getting the Source Code
of our dev manual: https://download.xcsoar.org/releases/6.8.13/XCSoar-developer-manual.pdf

If you need git for windows: https://gitforwindows.org

PR #296 (2020-05-03): Vagrant debian10
This updates the vagrant box to debian-10. We are also switching to the generic box provider from vagrantup. These boxes are built by lavabit and the entire [process](https://github.com/lavabit/robox/blob/master/http/generic.debian10.vagrant.cfg) is here on github.
The boxes support all possible providers for vagrant, such as: virtualbox,hyperv,libvirt and even parallels on OSX.

The clone process also now initializes submodules without which a build would fail.

Fixes #292 

I'm not entierly happy about adding the repos in the vagrantfile. This should be a seperate script in  [... cut]