XCSoar/XCSoar #295: Raspberry Pi 4 XCSoar does not run -> Failed dev/tty

elmiba, 2019-11-28:
Hi all,
I wanted to install XCSoar on my new RPi4 and it fails to start the program.
The error message is: 
`Warning: failed to set graphics mode on /dev/tty: Unpassender IOCTL (I/O-Control) für das Gerät`
`* failed to add service - alredy in use?`

_"Unpassender IOTCL für das Gerät" = Wrong or inappropriate IOCTL for the device_

Things I have tried:
* Reinstalling
*  older Versions (every Version from 6.8.8 to 6.8.10)
* updating OS
* creating rules file in udev (as it is instructed in the manual)
* turning off VNC (I thought this could be the problem)

I hope somebody can help me.

Regards

lordfolken, 2019-11-28:
Nobody validated xcsoar on an rpi4 so far.. Pi 1/2 or cubie are build targets but 4 not so far.. So you will have some experimentation to do. That being said, is that raspian is a relative generic platform, so it could work. 

From the [xcsoar manual](https://download.xcsoar.org/releases/6.8.11/XCSoar-manual.pdf):

```
On Raspberry Pi and Cubieboard

Install the Debian package as described above. However, unlike
on “regular” Linux, XCSoar will not use X11. Instead, it runs fullscreen on the Linux console.

XCSoar requires access to your input devices (/dev/input/event*).
By default, only root is granted access. To override this, create a
udev configuration file, e.g. /etc/udev/rules.d/99-input.rules:
KERNEL=="event*", NAME="input/%k", MODE="660", GROUP="input"

Create the group input and make your user a member:
groupadd input
adduser pi input
```

So XCSoar on PI accesses the Framebuffer directly. You cannot be running any graphical desktop on the PI in order to run it. 

So use the Raspbian Buster Lite variant from https://www.raspberrypi.org/downloads/raspbian/
this should anyway be a dedicated device for flying.

elmiba, 2019-12-02:
Thank you for your response.
I have not been successful yet, but the problem above went away when I closed the GUI of the RPi. 
Now it gets stuck when launching xcs from the terminal with out any hint on what is happening, but I will close this issue because my original problem got solved.

OscarKolsrud, 2020-02-20:
Hi @elmiba im facing the same issue. Have you gotten it to work?

Beerjockel, 2020-09-23:
I'm also trying to get XCSoar on my Pi 4 but got the same problems, switched to the Raspbian Lite version and run into the same problem as @elmiba . After starting xcsoar via console but nothing happens.

markus2marks, 2021-02-05:
the same problem here, has anyone found a solution for it?

lordfolken, 2021-05-27:
You can try strace xcsoar and see what blocks it.

MaxKellermann, 2021-07-01:
I cannot reproduce the problem, it works just fine. Since the submitter did not reply to Folken's request after a month, I'm closing this issue.

My theory is: he used the Pi1-3 package on a Pi4.

closed 2021-07-01 by MaxKellermann