XCSoar/XCSoar #302: ANDROID: missing correct development configuration with commits after 2018-07-19

ThHanika, 2019-12-12:
Hello masters, can you please setting the right install- and setting-scripts in 
/ide/provisioning/

I am only able to make a xcsoar-android.apk with the last 
commit e87d79bc324f9708eaf85f617aea28a606feeed5 from 2018-07-19

lordfolken, 2019-12-12:
i have recently updated the ide-provisioning scripts and the resulting vagrant box is able to build android.  
Have a look at #296

lordfolken, 2019-12-12:
can you describe your build environment?

ThHanika, 2019-12-12:
The System is Debian 9 (Stretch) a VM VirtualBox system.
also a host-system with xubuntu 18.04.1 using java8.
Installation of develope packets via /ide/provisioning.

lordfolken, 2019-12-17:
Can you try #296

lordfolken, 2020-05-04:
#296 should fix this. Otherwise reopen.

closed 2020-05-04 by lordfolken