XCSoar/XCSoar #303: Xcsoar latest release 6.8.11 not available for Raspberry Pi

montich, 2019-12-15:
I have tried to download the latest release Xcsoar 6.8.11 for Rasperry Pi from the download link on xcsoar.org. Unfortunately the link ends in a 404 not found error. Older versions can be downloaded.

Does anyone know, whether Xcsoar software will still be compiled and distributed for Raspberry Pi in the future?

lordfolken, 2020-04-20:
6.8.11 is missing. we now have 6.8.13 in the repo.

MaxKellermann, 2020-05-22:
I built 6.8.14 for Pi: https://download.xcsoar.org/releases/6.8.14/PI/
My Cubieboard is offline, and since the Pi3/Pi4, there's no reason anymore to use a Cubieboard. I don't want to support it anymore. Maybe there's a volunteer to do builds...

closed 2020-05-22 by MaxKellermann

kedder, 2020-05-22:
For the reference: XCSoar on Cubieboard is supported by Openvario project. However, even though it runs XCSoar on Cubieboard, it doesn't use CUBIE target anymore. The build target for OV is `UNIX`, and it uses standard OpenGL canvas (i.e. not `mali_native_window`). IOW, it doesn't rely on proprietary mali opengl implementation, but uses opensource one provided by mesa.

Code, guarded by `TARGET_HAS_MALI`  and `HAVE_MALI` defines is also not used for Openvario anymore.